stuartpittaway/diyBMSv4 #5: LM4040BIM3-2.5/NOPB Missing in BOM

phisp, 2019-09-07:
The Part LM4040BIM3-2.5/NOPB is Missing the BOM

Also some parts are mentioned double

R1 C17414 C84376
R2 	C95781
R2	C115316

And the pin headers

The BOM and the Schematic is missing for the ESPCircuit (pdf)

Thanks for your work

phisp, 2019-11-05:
Thanks for the remove of the double entries.
The part LM4040BIM3-2.0/NOPB or similar parts is missing within the BOM.

thaxy, 2019-12-28:
I cannot find a complete list of all required parts per module.
What is the easiest way to figure out what exactly is needed?
BOM is still incomplete, right? Is only the LM4040BIM3 missing?

phisp, 2019-12-29:
I have a complete list, with approx. prices at my branch.
As far as i can see the LM4040BIM3 is the only part missing. Use the 2.048V Type. The 2.5V Type is maybe causing a few issues on my boards.
Double check the components is always a good idea.

stuartpittaway, 2019-12-29:
Take a look in the JLCPCB branch of the code, there is a fuller BOM there including some parts for substitution (note there is a newer PCB using 0603 surface mount components in that branch)

closed 2020-01-20 by stuartpittaway