stuartpittaway/diyBMSv4 #6: doubled items in bom?

CrAzYDr1veR, 2019-10-13:
https://mega.nz/#!mUJDSYpQ!viS_HsleeJhOGl9Gnn1_PtPxnbJkhVMMld8XrH_PA4U

Ordering is a nightmare because lcsc has many items out of stock so while making a list i found some doubles is that normal?
I can follow instruction but get confused easily with that kind of stuff

v4.csv

R17, R18 | 2K2

R23 | 2K2

CellModules_LCSC_BillOfMaterials.xlsx

r1

C17414 | 0805W8F1002T5E | Uniroyal Elec | 0805 |10k

C84376 | RC0805FR-0710KL | YAGEO | 0805 |10k

r2

C115316 | RS-05K1001FT | Guangdong Fenghua Advanced Tech | 0805 | 1k

C95781 | RC0805FR-071KL | YAGEO | 0805 | 1k

R22 | 5W-10Ω missing?

stuartpittaway, 2019-10-14:
Thanks for the information, looks like a couple of duplicates crept in on the last update.

The last update was trying to work around LCSC items being discontinued.

Note that R22 (power resistor) is not required or recommended (optional component) use the surface mount resistors instead.

CrAzYDr1veR, 2019-10-18:
new bom items in red are not available on lcsc and have to be bought from othe places like mouser...
feel free to modify to your taste i see now how much work those lists require.
https://ufile.io/ec2kpm55

stuartpittaway, 2019-10-18:
There are alternative items from LCSC available that would also work - you can never keep a BOM list current as parts come and go over time.

CrAzYDr1veR, 2019-10-18:
true but having all items in one list make things a bit more easy