stuartpittaway/diyBMSv4 #8: [Kicad] Missing Library EL3H7-G.lib

hitech95, 2019-11-05:
I'm tring to open the project but looks like that the libraries are missing.
I'm new on Kicad so I might do something wrong.

Steps to reproduce:
1) Download & Install Kicad
2) Open the project
3) Tell Kicad to ad dthe Simbols to the default library
4) Open the schematic
5) Get an error

stuartpittaway, 2019-11-05:
Which version of kicad did you install?

hitech95, 2019-11-05:
I have installed Kicad 5.1.14

phisp, 2019-11-05:
I had a missing Libraries issues with an older version of Kicad too. I installed 5.1.4 and it worked fine.

hitech95, 2019-11-05:
> 
> 
> I had a missing Libraries issues with an older version of Kicad too. I installed 5.1.4 and it worked fine.

I probably need some default libray, that so far is not loaded. (Or at least I think)
I'm recreating the project on Altium so I can use a CAD that I know, but I would have preferred to use Kicad if the project was loading correctly.

phisp, 2019-11-05:
Go to Preferences -> Manage Symbol Libraries -> Project Specific Libraries
Delete si8631ab-b-is1 and EL3H7-G these parts are not needed anymore.
The EL3H7-G has probably been replaced by the HMHA2801.
The ESPCircuit needs the Wemos D1 Mini from https://github.com/rubienr/wemos-d1-mini-kicad
to be completed. However it is not need either.

hitech95, 2019-11-05:
> Go to Preferences -> Manage Symbol Libraries -> Project Specific Libraries
> Delete si8631ab-b-is1 and EL3H7-G these parts are not needed anymore.
> The EL3H7-G has probably been replaced by the [...]

Done but the schematic still have a lot of '???' instead of the symbols. Resistors, caps, all all marked with ?

hitech95, 2019-11-06:
Looks like I had a corrupted installation of KiCad!

closed 2019-11-06 by hitech95