stuartpittaway/diyBMSv4 #12: Can you add an "cpl" file?

ArchGalileu, 2019-12-03:
Hello

I would like to try the JLPCB SMT service

Missing .cpl file to proceed.

Tried with kicad but can't view components on 3d view, also on export off cpl (csv) missing 

`No header row found, please include the below info in the header row. Designator, Mid X, Mid Y, Layer, Rotation.`

Best regards
ArchGalileu

stuartpittaway, 2019-12-04:
Hello KICAD doesn't output CPL files, but it will produce a position file which should be similar.

I've uploaded it to here....

https://github.com/stuartpittaway/diyBMSv4/blob/master/Circuit/gerber/v4-top-pos.csv

atanisoft, 2019-12-05:
@stuartpittaway take a look at https://github.com/matthewlai/JLCKicadTools/tree/master/jlc_lib which can generate the CPL and BOM file for JLC.

ArchGalileu, 2019-12-05:
Will try ;)

Best regards 
ArchGalileu

closed 2020-01-02 by stuartpittaway