stuartpittaway/diyBMSv4 #15: Not reading all modules

mkranjcic, 2019-12-05:
In My configuration, I have 20 modules connected in chain, where first 16 are in bank0 and 4 in bank1.
At some point, data from bank1 stopped being read.
Green led flashes on bank0, and on last module in chain (4th in bank1) in circles.
All modules are in same room (current temp is 14 Celsius), and here you can see them showing different temperature and higher voltage:
[image]

Part of console log:
```
S:80/1/2241=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E6E9/Q:15 # R:80/81/2241=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =219F
Rules:000001111=N/F
S:80/3/2242=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9CCD/Q:14 # R:80/83/2242=3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700  =B76C
S:80/1/2243=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CE82/Q:15 # R:80/81/2243=F9B FBA FAB FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =9BA4
Rules:000001111=N/F
S:80/1/2244=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2A1A/Q:15 # R:80/81/2244=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =ED6C
S:80/3/2245=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7855/Q:14 # R:80/83/2245=3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700  =53F4
Rules:000001111=N/F
S:80/1/2246=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =271/Q:15 # R:80/81/2246=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =C507
S:80/3/2247=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =503E/Q:14 # R:80/83/2247=3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700  =7B9F

Rules:000001111=N/F
S:80/1/2248=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DB60/Q:15 # R:80/81/2248=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =1C16
S:80/1/2249=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4745/Q:15 # R:80/81/2249=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =8033
Rules:000001111=N/F
S:80/3/224A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3D61/Q:14 # R:80/83/224A=3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700  =16C0
S:80/1/224B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6F2E/Q:15 # R:80/81/224B=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =A858
Rules:000001111=N/F
S:80/1/224C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8BB6/Q:15 # R:80/81/224C=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =4CC0
S:80/3/224D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D9F9/Q:14 # R:80/83/224D=3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700  =F258
Rules:000001111=N/F
S:80/1/224E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A3DD/Q:15 # R:80/81/224E=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =64AB
S:80/3/224F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F192/Q:14 # R:80/83/224F=3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700  =DA33
Rules:000001111=N/F
S:80/1/2250=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =29B5/Q:15 # R:80/81/2250=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =EEC3
S:80/1/2251=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B590/Q:15 # R:80/81/2251=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =72E6
Rules:000001111=N/F
S:80/3/2252=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CFB4/Q:14 # R:80/83/2252=3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700  =E415
S:80/1/2253=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9DFB/Q:15 # R:80/81/2253=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =5A8D
Rules:000001111=N/F
S:80/1/2254=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7963/Q:15 # R:80/81/2254=F9B FBA FAF FA8 FB1 FA4 FAD FB4 FB2 FB1 F8E F8E F9B F9B F75 F83  =BE15
S:80/3/2255=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2B2C/Q:14 # R:80/83/2255=3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700 3700  =8D
Rules:000001111=N/F
```

Odd thing was that I have connected charger to batteries, and it did start to read from all modules, (this is on image, modules were dissipating power), with green light flashing rounding twice on bank0 + last module in bank1, and rounding twice on bank1. Later ti stopped again.

Let me know if I can somehow debug more.

I have reflashed all with version of firmware after you have fixed part with 16th module not being able to be in bank0.

Thank you

mkranjcic, 2019-12-30:
Hi,
I have captured data from restart of device until it stops reading all modules.
`state: 2 -> 3 (0)
                 pcf8574 not fitted
Connecting to WIFI
Connecting to Wi-Fi...
state: 3 -> 5 (10)
                  add 0
                       aid 1
                            cnt

                                connected with xxxxxx, channel 1
                                                                   dhcp client start...
                                                                                       scandone
                                                                                               ip:10.5.xxx.xxx,mask:255.255.255.0,gw:10.5.xxx.xxx
                                                                                                                                                Connected to Wi-Fi.
3. Connected IP:10.5.xxx.xxx
Requesting NTP from 10.5.xxx.xxx
Got NTP time: 18:47:30 30/12/2019
S:80/1/1=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925/Q:5 # Rules:000000111=N/F
R:80/81/1=EBE EF2 EC9 EEE EEA EDD EE2 F02 F00 EF8 E77 E80 E99 E8E E96 ECC  =AA1
S:80/3/2=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7301/Q:4 # R:80/83/2=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4200 3E00 3E00 3E00 3E00 3E00  =13DF
S:80/4/3=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CA8E/Q:7 # Rules:000001111=N/F
pm open,type:2 0
                R:80/84/3=109 0 E31 11E9 11EB 11EA AE3 11E7 0 10A8 0 0 C74 0 C5B 6E  =FC94
S:90/1/4=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7184/Q:6 # R:94/81/4=EE5 ED4 EFA EF6 0 0 0 0 0 0 0 0 0 0 0 0  =FC34
S:90/3/5=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =23CB/Q:9 # Rules:000001111=N/F
R:94/83/5=3E00 3E00 3E00 3E00 0 0 0 0 0 0 0 0 0 0 0 0  =CC6F
S:90/4/6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B22F/Q:12 # R:94/84/6=20C 20C 0 75C 0 0 0 0 0 0 0 0 0 0 0 0  =59F
S:80/1/7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7198/Q:11 # Rules:000001111=N/F
R:80/81/7=EBE EF2 EC9 EEE EEA EDD EE2 F06 F00 EF8 E77 E80 E99 E8E E96 EC7  =5E41
S:80/3/8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FAC6/Q:14 # R:80/83/8=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4200 3E00 3E00 3E00 3E00 3E00  =9A18
S:90/1/9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1CDB/Q:15 # Rules:000001111=N/F
R:94/81/9=EE5 ED4 EFA EF6 0 0 0 0 0 0 0 0 0 0 0 0  =916B
S:90/3/A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =66FF/Q:14 # R:94/83/A=3E00 3E00 3E00 3E00 0 0 0 0 0 0 0 0 0 0 0 0  =895B
S:80/1/B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =80E2/Q:15 # Rules:000001111=N/F
R:80/81/B=EBE EF2 EC9 EEE EEA EDD EE2 F06 F00 EF8 E77 E80 E99 E8E E96 EC7  =AF3B
S:80/3/C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AA10/Q:14 # R:80/83/C=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4200 3E00 3E00 3E00 3E00 3E00  =CACE
S:90/1/D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4C0D/Q:15 # Rules:000001111=N/F
R:94/81/D=EE5 ED4 EFA EF6 0 0 0 0 0 0 0 0 0 0 0 0  =C1BD
S:90/3/E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3629/Q:15 # R:94/83/E=3E00 3E00 3E00 3E00 0 0 0 0 0 0 0 0 0 0 0 0  =D98D
S:80/1/F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D034/Q:14 # Rules:000001111=N/F
R:80/81/F=EBE EF2 EC9 EEE EEA EDD EE2 F02 F00 EF8 E77 E80 E99 E8E E96 ECC  =D3B0
S:80/3/10=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =813/Q:15 # R:80/83/10=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4200 3E00 3E00 3E00 3E00 3E00  =68CD
S:90/1/11=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EE0E/Q:15 # Rules:000001111=N/F
R:94/81/11=EE5 ED4 EFA EF6 0 0 0 0 0 0 0 0 0 0 0 0  =63BE
S:90/3/12=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =942A/Q:14 # R:94/83/12=3E00 3E00 3E00 3E00 0 0 0 0 0 0 0 0 0 0 0 0  =7B8E
S:80/1/13=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7237/Q:15 # Rules:000001111=N/F
R:80/81/13=EBE EF2 EC9 EEE EEA EDD EE2 F02 F00 EF8 E77 E80 E99 E8E E96 ECC  =71B3
S:80/3/14=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =58C5/Q:14 # R:80/83/14=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4100 3E00 3E00 3E00 3E00 3E00  =40E1
S:90/1/15=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BED8/Q:15 # Rules:000001111=N/F
R:94/81/15=EE5 ED4 EFA EF6 0 0 0 0 0 0 0 0 0 0 0 0  =3368
S:90/3/16=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C4FC/Q:15 # R:94/83/16=3E00 3E00 3E00 3E00 0 0 0 0 0 0 0 0 0 0 0 0  =2B58
S:80/1/17=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =22E1/Q:14 # Rules:000001111=N/F
R:80/81/17=EBE EF2 EC9 EEE EEA EDD EE2 F06 F00 EF8 E77 E80 E99 E8E E96 ECC  =D1C2
S:80/3/18=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A9BF/Q:15 # R:80/83/18=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4100 3E00 3E00 3E00 3E00 3E00  =B19B
S:80/1/19=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FBF0/Q:15 # Rules:000001111=N/F
R:80/81/19=EBE EF2 EC9 EEE EEA EDD EE2 F02 F00 EF8 E77 E80 E99 E8E E96 ECC  =F874
S:80/3/1A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =81D4/Q:14 # R:80/83/1A=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4100 3E00 3E00 3E00 3E00 3E00  =99F0
S:80/1/1B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D39B/Q:15 # Rules:000001111=N/F
R:80/81/1B=EBE EF2 EC9 EEE EEA EDD EE2 F02 F00 EF8 E77 E80 E99 E8E E96 ECC  =D01F
S:80/3/1C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F969/Q:14 # R:80/83/1C=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4100 3E00 3E00 3E00 3E00 3E00  =E14D
S:80/1/1D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AB26/Q:15 # Rules:000001111=N/F
R:80/81/1D=EBE EF2 EC9 EEE EEA ED9 EE2 F06 F00 EF8 E77 E80 E99 E8E E96 ECC  =FF44
S:80/1/1E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1F68/Q:15 # R:80/81/1E=EBE EEE EC9 EEE EEA ED9 EE2 F06 F00 EF8 E77 E80 E99 E8E E96 ECC  =B67E
S:80/3/1F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4D27/Q:14 # Rules:000001111=N/F
R:80/83/1F=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4100 3E00 3E00 3E00 3E00 3E00  =5503
[... cut ...]
R:80/83/27=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 4100 3E00 3E00 3E00 3E00 3E00  =124
S:80/1/28=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925E/Q:15 # R:80/81/28=EBE EEE EC9 EEE EEA EDD EE2 F02 F00 EF8 E77 E80 E99 E92 E96 EC7  =A133
S:80/1/29=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E7B/Q:15 # Rules:000001111=N/F
R:80/81/29=EBE EF2 EC9 EEE EEA EDD EE2 F06 F00 EF8 E77 E80 E99 E8E E96 EC7  =21A2
S:80/3/2A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =745F/Q:14 # R
`
Is there something else I could try to capture?

stuartpittaway, 2019-12-31:
From the above log, you can see the S:80/1/ and S:90/1/ lines - this is a voltage reading command for bank 0 and then bank 1.

Bank 0 returns
R:80/81/1B=EBE EF2 EC9 EEE EEA EDD EE2 F02 F00 EF8 E77 E80 E99 E8E E96 ECC  =D01F

Bank 1 returns
R:94/81/15=EE5 ED4 EFA EF6 0 0 0 0 0 0 0 0 0 0 0 0  =3368

However after S:90/3/16 no further requests from the controller are made to bank 1.

Very strange and I suspect a coding error.

Can I ask a favour - can you reallocate the modules over the two banks, only have 15 modules on bank zero and 5 on bank one and see if that resolves this?

mkranjcic, 2019-12-31:
> Can I ask a favour - can you reallocate the modules over the two banks, only have 15 modules on bank zero and 5 on bank one and see if that resolves this?
> 
> It might also be worth checking the settings/configuration web page to see if the number of banks has dropped to 1 at this point as well.

Done, I have switched module 15 from bank0 to bank1, and console is below (after hard reset).
Observation - it takes very long to get configuration from module (minutes?). It is hard to tell when I can get it, and clicking all the time does not help (even more - looks like it does opposite).
Same thing after changing bank. If I reset exp too soon (after 30s-1m), it forgets that I have setup it this way.

```
state: 2 -> 3 (0)
                        pcf8574 not fitted
Connecting to WIFI
Connecting to Wi-Fi...
state: 3 -> 5 (10)
                  add 0
                       aid 1
                            cnt

                                connected with XXXX, channel 1
                                                                   dhcp client start...
                                                                                       scandone
                                                                                               ip:XXXX,mask:255.255.255.0,gw:XXXX
                                                                                                                                                Connected to Wi-Fi.
3. Connected IP:XXXX
Requesting NTP from XXXX
Got NTP time: 16:48:28 31/12/2019
S:80/1/1=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925/Q:5 # Rules:000000110=N/F
Rules:010000110=N/F
pm open,type:2 0
                S:80/3/2=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7301/Q:12 # Rules:010000110=N/F
S:80/4/3=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CA8E/Q:11 # S:90/1/4=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7184/Q:10 # S:90/3/5=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =23CB/Q:13 # S:90/4/6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B22F/Q:12 # S:80/1/7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7198/Q:11 # S:80/3/8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FAC6/Q:10 # S:90/1/9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1CDB/Q:9 # S:90/3/A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =66FF/Q:8 # S:80/1/B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =80E2/Q:7 # S:80/3/C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AA10/Q:6 # Rules:010000110=N/F
S:90/1/D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4C0D/Q:9 # S:90/3/E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3629/Q:8 # S:80/1/F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D034/Q:7 # S:80/3/10=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =813/Q:6 # S:90/1/11=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EE0E/Q:5 # R:95/81/D=FC1 FA0 F92 FD4 1FB5 0 0 0 0 0 0 0 0 0 0 0  =654C**FAILED PROCESS REPLY**

S:90/3/12=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =942A/Q:4 # S:80/1/13=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7237/Q:3 # R:8F/81/F=1F62 FE4 FBC FBE FCC FB1 FC3 FC2 FBF FBA FB7 FA4 FBA FC3 FD3 0  =D14A**FAILED PROCESS REPLY**

S:80/3/14=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =58C5/Q:2 # S:90/1/15=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BED8/Q:5 # S:90/3/16=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C4FC/Q:4 # Rules:010000110=N/F
S:80/1/17=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =22E1/Q:3 # S:80/3/18=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A9BF/Q:2 # S:90/1/19=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4FA2/Q:1 # S:90/3/1A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3586/Q:0 # R:8F/81/17=F62 FA4 FBC FBE FCC FB1 FC3 FC2 FBF FBA F97 FA4 FBA FBF FD3 0  =6DE0**FAILED PROCESS REPLY**

R:8F/83/18=3F00 4000 4B00 4000 4000 4000 4000 4000 3F00 3F00 3F00 3F00 4000 4000 4000 0  =5EDE**FAILED PROCESS REPLY**

S:80/1/1B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D39B/Q:3 # S:80/3/1C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F969/Q:2 # S:90/1/1D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1F74/Q:1 # S:90/3/1E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6550/Q:0 # Rules:010000110=N/F
R:8F/81/1B=F62 FA4 FBC FBE FCC FAD FC3 FC2 FBF FBF F97 FA4 FBA FBF FD3 0  =5EFA**FAILED PROCESS REPLY**

R:8F/83/1C=3F00 4000 4A00 4000 4000 4000 4000 4000 3F00 3F00 3F00 3F00 4000 4000 4000 0  =1B6E**FAILED PROCESS REPLY**

R:95/81/1D=FD3 FA0 F92 FD4 FB5 0 0 0 0 0 0 0 0 0 0 0  =7C**FAILED PROCESS REPLY**

R:95/83/1E=4000 4000 4000 4000 3F00 0 0 0 0 0 0 0 0 0 0 0  =203F
S:80/1/1F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =834D/Q:3 # R:8F/81/1F=F62 FA4 FBC FBE FCC FB1 FC3 FC2 FBF FBA F97 FA4 FBA FBF FD3 0  =CC4C
S:80/3/20=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FD98/Q:2 # Rules:000101110=N/F
R:8F/83/20=3F00 4000 4A00 4000 4000 4000 4000 4000 3F00 3F00 3F00 3F00 4000 4000 4000 0  =1F9F
S:90/1/21=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1B85/Q:5 # R:95/81/21=FC1 FA0 F92 FD4 FB5 0 0 0 0 0 0 0 0 0 0 0  =72C4
S:90/3/22=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =61A1/Q:4 # Rules:000001110=N/F
R:95/83/22=4000 4000 4000 4000 3F00 0 0 0 0 0 0 0 0 0 0 0  =24CE
S:80/1/23=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =87BC/Q:7 # R:8F/81/23=F62 FA4 FBC FBE FCC FAD FC3 FC2 FBF FBF F97 FA4 FBA FBF FD3 0  =ADD
S:80/3/24=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AD4E/Q:10 # Rules:000001110=N/F
R:8F/83/24=3F00 4000 4900 4000 4000 4000 4000 4000 3F00 3F00 3F00 3F00 4000 4000 4000 0  =70E3
S:90/1/25=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4B53/Q:9 # R:95/81/25=FC1 FA0 F92 FD4 FB5 0 0 0 0 0 0 0 0 0 0 0  =2212
S:90/3/26=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3177/Q:12 # Rules:000001110=N/F
R:95/83/26=4000 4000 4000 4000 3F00 0 0 0 0 0 0 0 0 0 0 0  =7418
S:80/1/27=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D76A/Q:15 # R:8F/81/27=F62 FA4 FBC FBE FCC FAD FC3 FC2 FBF FBA F97 FA4 FBA FBF FD3 0  =DCCF
S:80/3/28=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5C34/Q:14 # Rules:000001110=N/F
[... cut ...]
S:80/1/47=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2C5D/Q:15 # R:8F/81/47=F62 FA4 FBC FBE FCC FAD FC3 FC2 FBF FBA F97 FA4 FBA FBF FD3 0  =27F8
S:80/3/48=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A703/Q:14 # Rules:000001110=N/F
R:8F/83/48=3F00 4000 4500 4000 4000 3F00 3F00 4000 3F00 3E00 3F00 3F00 3F00 3F00 3F00 0  =26EF
S:80/1/49=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F54C/Q:15 # R:8F/81/49=F62 FA4 FBC FBE FCC FAD FC3 FC2 FBF FBA F97 FA0 FBA FBF FD3 0  =53DC
S:80/3/4A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8F68/Q:14 # Rules:000001110=N/F
R

```

stuartpittaway, 2020-01-20:
Sorry I didn't realise you had replied - whats the current state of your installation ?

mkranjcic, 2020-01-20:
Last time I checked - it was same. It did not help to have 15+5 or 16+4 combination.
I am preparing now full sized batteries (48p), and test was just on 3p, but I expect to have same situation. If needed, I can reconnect all back for testing.
Thx for help

mkranjcic, 2020-01-28:
So, a bit more debugging - it seams that send queue is being flooded, and not all packets are being sent, resulting in most cases that at beginning, when queue is not full - it works, but later it fills up, and packets for bank1 are dropped.
At first, increasing time for sending packets to 10s has helped, but now my colleague and I are looking to make async packet sending.
Will try to fix this and propose you to include in main code.

stuartpittaway, 2020-01-28:
The send queue will only get full if no replies are read. There's no need for async code in this context as the replies from the modules should be very consistent. Each module takes about 10ms to reply.

stuartpittaway, 2020-02-24:
Hello, the latest code has drastically changed the way data packets are sent and received, so I'm closing this issue - feel free to re-raise if the new code doesn't work for you.

closed 2020-02-24 by stuartpittaway