stuartpittaway/diyBMSv4 #16: Crash of controller with more then 12 cell modules, influxdb turned on and ext. temperature NTCs

wurstyboy, 2019-12-08:
After the last changes in the code in order to export 3 decimals for "v" to influxdb and also in order to send the external temperature to influx the controller hungs up after ca. 1 minute. This only happens with more than 12 cell modules (took me 2 hours to figure that out. I thought I had a hardware problem). After the hung up I cannot reach the interface anymore.
I guess that some timeout occurs. No data will be send to my db.
Maybe some buffer overflow or similar happens now after the changes.
(With the old code everything works fine, but ofcourse only 2 demecimal and no ext. temp.)

stuartpittaway, 2019-12-09:
Thanks for the bug report.  I think this is running out of buffer space when building up the string to send to influx.  I can fix this by reducing the variable names that are passed into influx but you will need to modify influx to receive them.

wurstyboy, 2019-12-09:
Thanks for your quick reply.

What about sending in two portions?

What changes are needen in chase of changing the variable names? Just the the names like "exttemp" and "intemp" to maybe "tex" and "tin"? Or do you mean a deeper setting in the database inself?

For now I will comment out the parameter internal temperature for influxdb export. I dont't need this number currently. Let's see if this runs...

commit 5f267e8c (2019-12-12): Issue #16 

To resolve #16 reduce parameter names in INFLUX request
closed 2019-12-12 by stuartpittaway