stuartpittaway/diyBMSv4 #18: Problems during charging

mkranjcic, 2019-12-30:
Hello,
While charging is turned on, controller does not read modules.
I have power of ESP connected from PC USB port, and 20S3P modules are connected to charger for this setup. Charger detects voltage difference and starts charging at 1A, at which point controller stops reading values from modules. Whole time leds on modules flash. 
At begining they flash as they do not get packets (randomly timeouts),than 4 leds are cashing each other for some time, than 2 leds are chasing each other all the time (first two over bank0, than 2 over bank1) with delay of cca 5-6 modules distance between green light.
After turning charger off, packets start to flow in single instance (no double green lights), and it captures both bank0 and bank1, and after some time, bank1 is excluded (reported separate bug for this)

Here is console capture:
```
R:80/81/13D=EBE EF2 EC9 EEE EEA ED9 EE2 F02 F00 EF8 E77 E80 E99 E8E E96 EC7  =F837
S:80/1/13E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3446/Q:15 # R:80/81/13E=EBE EF2 EC9 EEE EEA EDD EE2 F02 F00 EF8 E77 E80 E99 E8E E96 ECC  =37C2
S:80/3/13F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6609/Q:14 # Rules:000001111=N/F
R:80/83/13F=3E00 3E00 3E00 3E00 3E00 3E00 3D00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00  =3C8C
S:80/1/140=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4519/Q:15 # R:80/81/140=EBE EF2 EC9 EEE EEA EDD EE2 F02 F00 EF8 E7C E80 E99 E8E E96 EC7  =8450
S:80/1/141=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D93C/Q:15 # Rules:000001111=N/F
R:80/81/141=EBE EEE EC9 EEE EEA ED9 EE2 F02 F00 EF8 E7C E80 E99 E8E E96 EC7  =4240
S:80/3/142=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A318/Q:14 # R:80/83/142=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00  =8B9D
S:80/1/143=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F157/Q:15 # Rules:000001111=N/F
R:80/81/143=EBE EF2 EC9 EEE EEA ED9 EE2 F06 F00 EF8 E7C E80 E99 E8E E96 ECC  =BB02
S:80/3/144=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DBA5/Q:14 # Rules:010001111=N/F
====== Here is charging turned on ========
Rules:010001111=N/F
S:80/1/145=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =89EA/Q:15 # S:80/1/146=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3DA4/Q:14 # S:80/3/147=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6FEB/Q:13 # S:80/1/148=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E4B5/Q:12 # S:80/1/149=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7890/Q:15 # S:80/3/14A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2B4/Q:14 # Rules:010001111=N/F
S:80/1/14B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =50FB/Q:13 # S:80/3/14C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7A09/Q:12 # S:80/1/14D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2846/Q:11 # S:80/1/14E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9C08/Q:10 # S:80/3/14F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CE47/Q:9 # S:80/1/150=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1660/Q:8 # S:80/1/151=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8A45/Q:11 # S:80/3/152=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F061/Q:10 # S:80/1/153=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A22E/Q:9 # S:80/3/154=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =88DC/Q:8 # Rules:010001111=N/F
S:80/1/155=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DA93/Q:7 # S:80/3/156=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A0B7/Q:6 # S:90/1/157=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =46AA/Q:5 # S:90/3/158=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CDF4/Q:4 # S:80/1/159=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2BE9/Q:7 # S:80/3/15A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =51CD/Q:6 # S:90/1/15B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B7D0/Q:5 # S:90/3/15C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9D22/Q:4 # S:80/1/15D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7B3F/Q:3 # S:80/3/15E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =11B/Q:2 # Rules:010001111=N/F
S:90/1/15F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E706/Q:1 # S:90/3/160=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =99D3/Q:0 # S:80/1/161=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7FCE/Q:3 # S:80/3/162=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5EA/Q:2 # S:90/1/163=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E3F7/Q:1 # S:90/3/164=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C905/Q:0 # Rules:010001111=N/F
S:80/1/165=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2F18/Q:3 # S:80/3/166=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =553C/Q:2 # S:90/1/167=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B321/Q:1 # S:90/3/168=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =387F/Q:0 # S:80/1/169=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DE62/Q:3 # S:80/3/16A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A446/Q:2 # Rules:010001111=N/F
S:90/1/16B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =425B/Q:1 # S:90/3/16C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =68A9/Q:0 # S:80/1/16D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8EB4/Q:3 # S:80/3/16E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F490/Q:2 # S:90/1/16F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =128D/Q:1 # S:90/3/170=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CAAA/Q:0 # Rules:010001111=N/F
S:80/1/171=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2CB7/Q:3 # S:80/3/172=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5693/Q:2 # S:90/1/173=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B08E/Q:1 # S:90/3/174=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9A7C/Q:0 # Rules:010001111=N/F
S:80/1/175=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7C61/Q:3 # S:80/3/176=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =645/Q:2 # S:90/1/177=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E058/Q:1 # S:90/3/178=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6B06/Q:0 # Rules:010001111=N/F
S:80/1/179=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8D1B/Q:3 # S:80/3/17A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F73F/Q:2 # S:90/1/17B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1122/Q:1 # S:90/3/17C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3BD0/Q:0 # S:80/1/17D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DDCD/Q:3 # S:80/3/17E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A7E9/Q:2 # Rules:010001111=N/F
S:90/1/17F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =41F4/Q:1 # S:90/3/180=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D96E/Q:0 # S:80/1/181=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3F73/Q:3 # S:80/3/182=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4557/Q:2 # S:90/1/183=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A34A/Q:1 # S:90/3/184=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =89B8/Q:0 # Rules:010001111=N/F
S:80/1/185=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6FA5/Q:3 # S:80/3/186=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1581/Q:2 # S:90/1/187=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F39C/Q:1 # S:90/3/188=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =78C2/Q:0 # Rules:010001111=N/F
S:80/1/189=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9EDF/Q:3 # S:80/3/18A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E4FB/Q:2 # S:90/1/18B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2E6/Q:1 # S:90/3/18C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2814/Q:0 # Rules:010001111=N/F
S:80/1/18D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CE09/Q:3 # S:80/3/18E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B42D/Q:2 # S:90/1/18F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5230/Q:1 # S:90/3/190=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8A17/Q:0 # S:80/1/191=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6C0A/Q:3 # S:80/3/192=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =162E/Q:2 # Rules:010001111=N/F
S:90/1/193=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F033/Q:1 # S:90/3/194=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DAC1/Q:0 # S:80/1/195=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3CDC/Q:3 # S:80/3/196=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =46F8/Q:2 # S:90/1/197=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A0E5/Q:1 # S:90/3/198=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2BBB/Q:0 # Rules:010001111=N/F
S:80/1/199=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CDA6/Q:3 # S:80/3/19A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B782/Q:2 # S:90/1/19B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =519F/Q:1 # S:90/3/19C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7B6D/Q:0 # Rules:010001111=N/F
S:80/1/19D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9D70/Q:3 # S:80/3/19E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E754/Q:2 # S:90/1/19F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =149/Q:1 # S:90/3/1A0=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7F9C/Q:0 # Rules:010001111=N/F
S:80/1/1A1=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9981/Q:3 # S:80/3/1A2=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E3A5/Q:2 # S:90/1/1A3=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5B8/Q:1 # S:90/3/1A4=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2F4A/Q:0 # S:80/1/1A5=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C957/Q:3 # S:80/3/1A6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B373/Q:2 # Rules:010001111=N/F
S:90/1/1A7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =556E/Q:1 # S:90/3/1A8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DE30/Q:0 # S:80/1/1A9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =382D/Q:3 # S:80/3/1AA=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4209/Q:2 # S:90/1/1AB=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A414/Q:1 # S:90/3/1AC=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8EE6/Q:0 # Rules:010001111=N/F
S:80/1/1AD=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =68FB/Q:3 # S:80/3/1AE=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =12DF/Q:2 # S:90/1/1AF=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F4C2/Q:1 # S:90/3/1B0=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2CE5/Q:0 # Rules:010001111=N/F
[... cut ...]
S:80/3/212=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2BE1/Q:14 # R:80/83/212=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00  =364
Rules:000001111=N/F
S:80/1/213=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =79AE/Q:15 # R:80/81/213=EEF F2C EF5 F11 F0E F05 F09 F2A F1F F24 E9F EA4 EBD EB1 EC3 EF4  =D6C1
S:80/3/214=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =535C/Q:14 # R:80/83/214=3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00 3E00  =7BD9
Rules:000001111=N/F
S:80/1/215=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =113/Q:15 # R
```

(here is also visible at the end of log how controller starts skipping reading from bank1)

stuartpittaway, 2019-12-30:
Thanks for the traces.

Once charging starts the controller never receives a single reply, you mention that the lights chase each other for a little while.

Is it possible for you to disconnect all but 1 module so the controller + 1 module are connected and try charging again - does this work?

If it does increase the modules one by one until the bank 0 is complete and hopefully working.

We can move on from then.

mkranjcic, 2019-12-31:
Hello,
Here is with just one module (but still 2 active banks in config)
```
Rules:000000110=N/F
S:80/1/441E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6B5B/Q:3 # R:81/81/441E=F2D 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2A7C
S:80/3/441F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3914/Q:2 # R:81/83/441F=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8F4E
S:90/1/4420=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F393/Q:1 # R:90/1/4420=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F393**FAILED PROCESS REPLY**

S:90/3/4421=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A1DC/Q:0 # R:90/3/4421=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A1DC**FAILED PROCESS REPLY**

S:80/1/4422=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6FAA/Q:3 # R:81/81/4422=F2D 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2E8D
S:80/3/4423=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3DE5/Q:2 # R:81/83/4423=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8BBF
Rules:000000110=N/F
S:90/1/4424=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A345/Q:1 # R:90/1/4424=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A345**FAILED PROCESS REPLY**

S:90/3/4425=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F10A/Q:0 # R:90/3/4425=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F10A**FAILED PROCESS REPLY**

S:80/1/4426=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3F7C/Q:3 # R:81/81/4426=F2D 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7E5B
S:80/3/4427=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6D33/Q:2 # R:81/83/4427=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DB69
S:90/1/4428=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =523F/Q:1 # R:90/1/4428=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =523F**FAILED PROCESS REPLY**

S:90/3/4429=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =70/Q:0 # R:90/3/4429=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =70**FAILED PROCESS REPLY**

Rules:000000110=N/F
S:80/1/442A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CE06/Q:3 # R:81/81/442A=F2D 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8F21
S:80/3/442B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9C49/Q:2 # R:81/83/442B=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2A13
S:90/1/442C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2E9/Q:1 # R:90/1/442C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2E9**FAILED PROCESS REPLY**

S:90/3/442D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =50A6/Q:0 # R:90/3/442D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =50A6**FAILED PROCESS REPLY**

Rules:000000110=N/F
S:80/1/442E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9ED0/Q:3 # Rules:010000110=N/F
Rules:010000110=N/F
S:80/3/442F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CC9F/Q:10 # S:90/1/4430=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A0EA/Q:9 # S:90/3/4431=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F2A5/Q:8 # S:80/1/4432=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3CD3/Q:11 # S:80/3/4433=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6E9C/Q:10 # S:90/1/4434=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F03C/Q:9 # S:90/3/4435=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A273/Q:8 # Rules:010000110=N/F
S:80/1/4436=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6C05/Q:7 # S:80/3/4437=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3E4A/Q:6 # S:90/1/4438=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =146/Q:5 # S:90/3/4439=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5309/Q:4 # S:80/1/443A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9D7F/Q:7 # S:80/3/443B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CF30/Q:6 # S:90/1/443C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5190/Q:5 # S:90/3/443D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3DF/Q:4 # S:80/1/443E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CDA9/Q:3 # S:80/3/443F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9FE6/Q:2 # Rules:010000110=N/F
S:90/1/4440=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8A4/Q:1 # S:90/3/4441=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5AEB/Q:0 # S:80/1/4442=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =949D/Q:3 # S:80/3/4443=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C6D2/Q:2 # S:90/1/4444=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5872/Q:1 # S:90/3/4445=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A3D/Q:0 # Rules:010000110=N/F
S:80/1/4446=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C44B/Q:3 # S:80/3/4447=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9604/Q:2 # S:90/1/4448=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A908/Q:1 # S:90/3/4449=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FB47/Q:0 # S:80/1/444A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3531/Q:3 # S:80/3/444B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =677E/Q:2 # Rules:010000110=N/F
S:90/1/444C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F9DE/Q:1 # S:90/3/444D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AB91/Q:0 # S:80/1/444E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =65E7/Q:3 # S:80/3/444F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =37A8/Q:2 # S:90/1/4450=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5BDD/Q:1 # S:90/3/4451=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =992/Q:0 # Rules:010000110=N/F
S:80/1/4452=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C7E4/Q:3 # S:80/3/4453=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =95AB/Q:2 # S:90/1/4454=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B0B/Q:1 # S:90/3/4455=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5944/Q:0 # Rules:010000110=N/F
S:80/1/4456=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9732/Q:3 # S:80/3/4457=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C57D/Q:2 # S:90/1/4458=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FA71/Q:1 # S:90/3/4459=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A83E/Q:0 # Rules:010000110=N/F
S:80/1/445A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6648/Q:3 # S:80/3/445B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3407/Q:2 # S:90/1/445C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AAA7/Q:1 # S:90/3/445D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F8E8/Q:0 # S:80/1/445E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =369E/Q:3 # S:80/3/445F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =64D1/Q:2 # Rules:010000110=N/F
S:90/1/4460=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AE56/Q:1 # S:90/3/4461=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FC19/Q:0 # S:80/1/4462=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =326F/Q:3 # S:80/3/4463=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6020/Q:2 # S:90/1/4464=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FE80/Q:1 # S:90/3/4465=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =ACCF/Q:0 # Rule [... cut ...]
Rules:000000110=N/F
S:80/1/44A6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =84F6/Q:3 # R:81/81/44A6=F5D 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BBC8
S:80/3/44A7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D6B9/Q:2 # R:81/83/44A7=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =60E3
S:90/1/44A8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E9B5/Q:1 # R:90/1/44A8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E9B5**FAILED PROCESS REPLY**

S:90/3/44A9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BBFA/Q:0 # R:90/3/44A9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BBFA**FAILED PROCESS REPLY**

```

stuartpittaway, 2019-12-31:
Is this with the charger on or off?  Can you set the number of banks to 1 in the config (you can easily change it back afterwards).

What are you using on the controller side - are you using one of my circuit boards?

It feels like a problem with the optoisolator on the controller - can you double check the soldering on that (perhaps reflow some of the points).

stuartpittaway, 2019-12-31:
By the way the "**FAILED PROCESS REPLY**" messages above are a result of having 2 banks configured but only 1 bank actually connected - it indicates that the controller sent a request and a reply was received but no module processed the command.

mkranjcic, 2019-12-31:
> Is this with the charger on or off? Can you set the number of banks to 1 in the config (you can easily change it back afterwards).

Ok, below is new capture with marking when charger is on and off. Also, when I have switched to only one bank.

> 
> What are you using on the controller side - are you using one of my circuit boards?

No, I use NodeMCU with breadboard and jumper wires. Optocoupler is soldered to pins and used "in air".

> It feels like a problem with the optoisolator on the controller - can you double check the soldering on that (perhaps reflow some of the points).

Ok, I will re-solder optocoupler and try again.

```
=========Charger is off==========
S:80/1/5F3C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =112B/Q:3 # R:81/81/5F3C=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7C73
S:80/3/5F3D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4364/Q:2 # R:81/83/5F3D=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F53E
S:90/1/5F3E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8D12/Q:1 # R:90/1/5F3E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8D12**FAILED PROCESS REPLY**

S:90/3/5F3F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DF5D/Q:0 # R:90/3/5F3F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DF5D**FAILED PROCESS REPLY**

Rules:000000110=N/F
S:80/1/5F40=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =481F/Q:3 # R:81/81/5F40=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2547
S:80/3/5F41=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1A50/Q:2 # R:81/83/5F41=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AC0A
S:90/1/5F42=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D426/Q:1 # R:90/1/5F42=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D426**FAILED PROCESS REPLY**

S:90/3/5F43=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8669/Q:0 # R:90/3/5F43=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8669**FAILED PROCESS REPLY**

Rules:000000110=N/F
S:80/1/5F44=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =18C9/Q:3 # R:81/81/5F44=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7591
S:80/3/5F45=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4A86/Q:2 # R:81/83/5F45=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FCDC
S:90/1/5F46=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =84F0/Q:1 # R:90/1/5F46=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =84F0**FAILED PROCESS REPLY**

S:90/3/5F47=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D6BF/Q:0 # R:90/3/5F47=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D6BF**FAILED PROCESS REPLY**
============= Settings set to just 1 bank=============
Rules:000000110=N/F
S:80/1/5F48=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E9B3/Q:1 # R:81/81/5F48=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =84EB
S:80/3/5F49=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BBFC/Q:0 # R:81/83/5F49=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DA6
S:80/1/5F4A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C1D8/Q:1 # R:81/81/5F4A=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =AC80
S:80/3/5F4B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9397/Q:0 # R:81/83/5F4B=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =25CD
Rules:000000110=N/F
S:80/1/5F4C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B965/Q:1 # R:81/81/5F4C=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D43D
S:80/3/5F4D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EB2A/Q:0 # R:81/83/5F4D=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5D70
Rules:000000110=N/F
S:80/1/5F4E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =910E/Q:1 # R:81/81/5F4E=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FC56
S:80/3/5F4F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C341/Q:0 # R:81/83/5F4F=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =751B
Rules:000000110=N/F
S:80/1/5F50=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1B66/Q:1 # R:81/81/5F50=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =763E
S:80/3/5F51=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4929/Q:0 # R:81/83/5F51=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =FF73
Rules:000000110=N/F
S:80/1/5F52=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =330D/Q:1 # R:81/81/5F52=F28 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =5E55
S:80/3/5F53=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6142/Q:0 # R:81/83/5F53=3D00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =D718
=========Charger turned on=========
S:80/1/5F54=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4BB0/Q:1 # Rules:000000110=N/F
S:80/3/5F55=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =19FF/Q:2 # Rules:010000110=N/F
S:80/1/5F56=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =63DB/Q:1 # S:80/3/5F57=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3194/Q:0 # S:80/1/5F58=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BACA/Q:1 # S:80/3/5F59=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E885/Q:0 # Rules:010000110=N/F
S:80/1/5F5A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =92A1/Q:1 # S:80/3/5F5B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C0EE/Q:0 # Rules:010000110=N/F
S:80/1/5F5C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EA1C/Q:1 # S:80/3/5F5D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =B853/Q:0 # S:80/1/5F5E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C277/Q:1 # S:80/3/5F5F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9038/Q:0 # Rules:010000110=N/F
S:80/1/5F60=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EEED/Q:1 # S:80/3/5F61=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BCA2/Q:0 # Rules:010000110=N/F
S:80/1/5F62=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C686/Q:1 # S:80/3/5F63=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =94C9/Q:0 # Rules:010000110=N/F
S:80/1/5F64=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BE3B/Q:1 # S:80/3/5F65=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EC74/Q:0 # Rules:010000110=N/F
S:80/1/5F66=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9650/Q:1 # S:80/3/5F67=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C41F/Q:0 # S:80/1/5F68=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4F41/Q:1 # S:80/3/5F69=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1D0E/Q:0 # Rules:010000110=N/F
S:80/1/5F6A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =672A/Q:1 # S:80/3/5F6B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3565/Q:0 # Rules:010000110=N/F
S:80/1/5F6C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1F97/Q:1 # S:80/3/5F6D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4DD8/Q:0 # Rules:010000110=N/F
[... cut ...]
S:80/3/5FF5=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =487/Q:0 # R:81/83/5FF5=6F00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4377
S:80/1/5FF6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7EA3/Q:1 # R:81/81/5FF6=F4C 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DC1E
S:80/3/5FF7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2CEC/Q:0 # R:81/83/5FF7=6E00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E535
Rules:000000110=N/F
S:80/1/5FF8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =A7B2/Q:1 # R:81/81/5FF8=F4C 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =50F
S:80/3/5FF9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F5FD/Q:0 # R:81/83/5FF9=6C00 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3057
Rules:000000110=N/F

```

stuartpittaway, 2019-12-31:
> No, I use NodeMCU with breadboard and jumper wires. Optocoupler is soldered to pins and used "in air".

Breadboard and jumper wires wires will act like an aerial and pick up all sorts of noise and corrupt the data signals.

I can see 1 reply was received during the charging and this was heavly corrupted.

S:80/1/5F8E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7741/Q:1 

R:F8/B8/A1F6=F07F A0F0 50E0 5078 F0F0 F050 E0E0 D0E0 A0A0 E0E0 50F0 E0F0 F0E0 A0A0 F0E0 B3F8  =302 **FAILED PROCESS REPLY**

Are you using the same Optocoupler chip as I used in the modules - HMHA2801 ?

mkranjcic, 2019-12-31:
> Breadboard and jumper wires wires will act like an aerial and pick up all sorts of noise and corrupt the data signals.

I know this, but since those are not too fast busses - I was hoping it will work.
I did not order wemos boards - since I have a lot of nodemcu-s at home.

> Are you using the same Optocoupler chip as I used in the modules - HMHA2801 ?

Yes, same optocoupler. I have now reduced number of jumper wires - it did not help :(
It is interesting that packets are streaming "ok" over all boards, and than nodemcu can not read it, while on other hand - when it is not charging - it works.

stuartpittaway, 2019-12-31:
You might be able to alter the resistor values connecting to the optocoupler and see if they helps remove the noise.

mkranjcic, 2019-12-31:
> You might be able to alter the resistor values connecting to the optocoupler and see if they helps remove the noise.

I will try to do it. Any suggestions?
Now I have 3k as pulldown on Receive and 220 on Transmit/optocoupler.

stuartpittaway, 2020-01-01:
I would increase the value of the 3k resistor - i'm using 4k7 but if you have something similar or even just a 10k resistor I would try that.

mkranjcic, 2020-01-04:
I have tried different values for resistor, and it was same behaviour.
Than I got some cheap oscilloscope (bitscope), connected async0(digital) and channelA(analog) to optocoupler and here is without charging:
[image]

And when I turn charger on (3k resistor):
[image]

And 10k with charger on:
[image]

And 1k with charger on:
[image]

Also, what I notice is that when charger is on - USB devices (controller, usb2 serial and oscilloscope via hub) are disconnecting periodically. 
I will need to check if there is some looping over charger and mains...

stuartpittaway, 2020-01-06:
Hi, definately something weird going on if you are also getting disconnection on the USB devices - is the controller module powered by the PC?  

Do you have an external battery powerbank you can power the controller from to isolate it and see if the noise goes away?

wurstyboy, 2020-01-19:
By the way: I lose conection to all modules when I have current flow (charge and discharge) around >15 A. At around 10 A CRC errors occur. My setup it pretty tight. The battery main cable is only 10-20 cm away from the esp controller module. The cables cross very close to another also. So I guess I have a noise problem. Next I want to strecht everything a bit. Lets see...

stuartpittaway, 2020-01-20:
Do you have an oscilloscope ?  I expect there is AC noise getting into the system and affecting the controller/modules.

mkranjcic, 2020-01-28:
> Hi, definately something weird going on if you are also getting disconnection on the USB devices - is the controller module powered by the PC?
> 
> Do you have an external battery powerbank you can power the controller from to isolate it and see if the noise goes away?

Yes, controller module is powered from PC. I will check to power it via battery/powerbank and retest. 
Will retest and post results.

Unfortunately - I do not have good oscilloscope, but this "poor man" one.
Also, I still do not have all connected - so I can not test full current charge/discharge situation. For me, this is happening in less than 1-2A charging...

stuartpittaway, 2020-02-16:
How is this setup working now?  Did you get any improvement?

stuartpittaway, 2020-02-24:
Hello, the latest code has drastically changed the way data packets are sent and received, so I'm closing this issue - feel free to re-raise if the new code doesn't work for you.

closed 2020-02-24 by stuartpittaway