stuartpittaway/diyBMSv4 #21: Cell module brownouts

delboy711, 2020-01-08:
Two out of my twelve cell modules suffered with brownouts during bypass. Symptoms were,  communication errors, sluggish or no response to commands, and EEPROM corruption.  The corrupted EEPROM  would have a good CRC implying RAM was corrupted and then written to EEPROM. I know the CRC must be good, because on the next power up it did not load default config and all parameters were random values.
The hardware was carefully checked, and fuses and flash memory reburned, and EEPROM cleared,  with no change in behaviour. On starting a bypass it would fail again.

The brownouts were resolved by;
A/ Increasing the BOD threshold to 2.7V - This stopped the EEPROM corruption, but continual brownout resets during bypass made the module still unusable.

B/ Increasing the load resistance by inserting an additional 2R2 in each leg of the chain, making a total resistance of 5R5. This worked, but of course will reduce the effectiveness of bypass.

C/ Commoning the centre point of the 2 load resistor chains together and adding a 10uF ceramic cap from there to ground. This smooths off the rising and falling edges of the PWM, and worked effectively.

I settled on using solution 'C' as well as removing the line  'Settings::WriteConfigToEEPROM((uint8_t*)&myConfig, sizeof(myConfig),EEPROM_CONFIG_ADDRESS);' in the setup function. This statement is unnecessary since it is just writing default config, and during a brownout there is a potential this statement could be randomly executed to write bad data with a good CRC into the EEPROM which is the only way I can explain how the EEPROM got corrupted yet still had a good CRC.

delboy711, 2020-01-08:
I should add, that I am using my own layout for the cell module PCBs. The circuit is identical to Stuart's apart from the addition of a 3A fuse at the power input, and a 100nF bypass capacitor between the MCU power pins.

stuartpittaway, 2020-01-09:
Hi Delboy711, thanks for the information.

What cells are you using and voltage range that you are dragging the cells below 2.7V ?

My existing circuit already has a 100nF bypass cap over the MCU - thats component C1

delboy711, 2020-01-09:
@stuartpittaway  My cell packs are a mixture of reclaimed 18650s. I noticed the problem when connecting a cell module to a pack at 4150mV with bypass threshold set to 4100mV. I also see the same effect when attached to my bench power supply.
During bypass the cell voltage does not drop more than around 50mV, and looking on my scope I have not been able to capture any spikes below the BOD threshold. Yet the MCU does go into lah-lah land as soon as bypass commences.

stuartpittaway, 2020-01-10:
This only happens on 2 of the modules or all of them?  I also use a bench power supply for testing and haven't got it to brownout or crash like this

delboy711, 2020-01-10:
Just the two.  Some parts may be more sensitive to transients on the power rail than others.
In terms of what action you might consider I suggest removing the WriteConfigToEEPROM() statement in setup since it does not do anything useful and would save a few bytes.

I did wonder if Issue #18   might be related. Transient noise from the OP's charger could be affecting the MCUs.

stuartpittaway, 2020-01-10:
Interesting stuff. #18 may be related however that issue seems to also have larger problems like usb drop outs and mains noise on the power lines.

stuartpittaway, 2020-01-12:
Hi @delboy711 on the capacitor fix you put on the circuit, is this simply a 10uF between the positive side of the resistors and gnd ?

delboy711, 2020-01-13:
@stuartpittaway  No. I  joined the ground side of R8 and R9 together, and put 10uF between there and ground. The idea being that there would be an RC time constant between the MCU power pin and the cap on both rising and falling edges as the MOSFET switches. It may be unnecessary to do it that way, but it worked so I stopped looking further.

stuartpittaway, 2020-01-13:
I'm probably being dim here but R8 and R9 are in the middle of the resistor string (one side has R6+R8+R10+R12 the other R7+R9+R11+R13) by joining r8+r9 together you will alter the load resistance.  With the capacitor in place, wouldn't this always be charged from the VCC side of the resistors and the ground from the capacitor ?

delboy711, 2020-01-14:
@stuartpittaway  The resistance is unchanged. Since the resistors are all the same value those two points are both at the same potential, Vcc/2.
When the FET is switched off the cap will charge to Vcc, and when switched on it will discharge down to Vcc/2. Either way there is a time delay before the voltage settles which results in a slower change in voltage at the MCU power pin. 
(In use, I observe that the cell voltage drops about 100mV when bypass is active.)

stuartpittaway, 2020-01-14:
Thanks for that, just for complete understanding, I've drawn a really crude picture, is this correct?  Its probably worth me copying the solution if its helped so much.

[image]

delboy711, 2020-01-15:
Yes. That is it.

phisp, 2020-09-14:
I have also issues with the B.O.D. set to 2.5V. The board measures a higher voltage and begins discharging until the battery block is completly drained. I will set the B.O.D. up to 3.0V and see if the issues resolve, as the balancing should stop at low voltages anyway.

The other pack has 20P3S and a voltage drop below the 2.5V is more unlikely. The 3P13S Pack has a very high likelyhood for package errors.

DanoPTT, 2020-09-27:
Measuring of higher voltage (for real low voltage) is the result of the use of improper voltage reference it this case. For such low voltage is better to use something like REF3020, which work with voltage drop just few mV (It is 3pin voltage reference, not shunt type).

stuartpittaway, 2020-09-28:
> REF3020

Hi DanoPTT, how would the change in voltage reference help in this situation?  I've not seen a problem with the reference voltage on the modules as they currently stand.

The 4.21 design also uses a lower voltage reference 1.25V - using https://datasheet.lcsc.com/szlcsc/1808081832_UTC-Unisonic-Tech-TL432G-AE3-R_C171461.pdf

phisp, 2021-10-16:
Thanks.
I used the higher voltage reference LM4040BIM3-2.5/NOPB 2.5V, due to ordering error and used 2x 13 modules each in parallel on a power supply with a battery to smooth the voltage. The difference occours at voltages below 3200mV on the measurements. I will try a 1.25V reference next.

[image]
[image]

closed 2022-04-14 by stuartpittaway