stuartpittaway/diyBMSv4 #24: Display

abdulhakim-genco, 2020-01-22:
In order to monitor cells voltage and temperature it would be nice to have small display shows total battery voltage, max cell temperature and min&max cell voltage.
I am planing to connect a TFT display to ESP8266 using TXD0 and RXD0 pins, but my knowledge with programming is very limited can you show me where should I make changes to export those values via serial port
thanks

stuartpittaway, 2020-01-22:
Hello, you should be able to connect a display to the existing ESP controller (you don't need a second one) using the i2c connections on the controller board.  I've not looked into it myself but should be easy enough.

closed 2020-02-26 by stuartpittaway

gen-ben, 2020-02-29:
hello, thanks for your reply, yes i will use the existing ESP but I meant that could I use the free Serial port with TX and RX labels on wemos board to communicate with TFT display or other microcontroller?.
you are using Serial and Serial1 in your code is there any way to add another serial function to control the third one.

[image]