stuartpittaway/diyBMSv4 #26: Internal temperature reading problem (when splitting packs in multiple banks)

Klauzen, 2020-02-07:
I've also a problem to report, by splitting the packs in to banks (bank0 and bank1).

The internal temperature on bank0 is updating permanent perfectly, while the bank1 is not updating the temperature. When I restart the controller, the current temperatures of both banks are listed correct on the webpage. Afterwards, only the bank0 is updating!

stuartpittaway, 2020-02-07:
Thanks for reporting, I'll try and take a look over the weekend.

mkranjcic, 2020-02-07:
This is same problem that I have with 20 modules. My conclusion is that queue is overfilled with messages, and messages for bank1 are skipped while for bank0 fit into queue.
It seams like generating messages and putting into queue is faster than emptying queue. 
It has helped when timer for generating messages was increased to 12s - for my setup.

Klauzen, 2020-02-08:
Ah okay! Maybe this also help for investigating the problem: Since I updated the code to the newest version of every cell and the controller board (I didn't updated since november 2019), the green LED's doesn't blink trough in one passing. Much more, bank0 is blinking trough 2...4 times and after a short break the bank1 is following with only one cycle... then, it repeats.

stuartpittaway, 2020-02-09:
Hi, I've updated the controller code to add some debug timings so we can diagnose this problem better.

You will need to connect a serial terminal to the 2 debug pins on the controller, the serial output will now include a "T" and "E" value.  The T is an indication of the milliseconds taken between the request going out and the reply being received.  The E is an internal counter, which could indicate queue jams.

`S:80/1/23=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =87BC/Q:3 #  T:470 E:0 R:82/81/23=F02 F02 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =476B
`

The above example is 470ms which is 4 modules configured in 2 banks, although 4 modules in 1 bank produces about the same timing.

Thats about 117ms for each module to process the packet and transmit the data.  20 modules should take 2340milliseconds - or about 2.5 seconds per request to pass through all 20 modules.

For 2 banks, we make a request every 4 seconds which then queues 4 messages (timerEnqueueCallback)

Every half second, we test for replies (timerTransmitCallback) an error is generated if we have not received a reply after 20 calls to this routine (about 10 seconds)

Note that the code has now moved to its own repo
https://github.com/stuartpittaway/diyBMSv4Code

Klauzen, 2020-02-15:
Hey Stuart,
thanks! After building and uploading the required code to the ESP, here is the console output:
`Found pcf8574
Clear AP settings1
Setup Access Point
mode : sta(84:f3:eb:e4:6f:03)
add if0
scandone
del if0
usl
mode : softAP(86:f3:eb:e4:6f:03)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
S:80/1/1=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925/Q:2 # Rules:000000100=
 T:1411 E:2 R:87/81/1=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =B891
S:80/3/2=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7301/Q:1 #  T:1410 E:2 R:87/83/2=3332 3332 3332 3332 3332 3332 3331 0 0 0 0 0 0 0 0 0  =47CB
S:80/4/3=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CA8E/Q:0 #  T:1399 E:2 R:87/84/3=0 0 2 0 0 0 2 0 0 0 0 0 0 0 0 0  =AAB7
S:80/1/4=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C5D6/Q:1 #  T:1405 E:2 R:87/81/4=FE6 FE2 FE6 FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =F627
Rules:000001000=
S:80/3/5=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9799/Q:0 #  T:1410 E:2 R:87/83/5=3332 3332 3332 3332 3332 3332 3331 0 0 0 0 0 0 0 0 0  =A353
S:80/1/6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EDBD/Q:1 #  T:1404 E:2 R:87/81/6=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =5C09
S:80/3/7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BFF2/Q:0 #  T:1410 E:2 R:87/83/7=3332 3332 3332 3332 3332 3332 3331 0 0 0 0 0 0 0 0 0  =8B38
Rules:000001000=
S:80/1/8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =34AC/Q:1 #  T:1404 E:2 R:87/81/8=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =8518
S:80/3/9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =66E3/Q:0 #  T:1410 E:2 R:87/83/9=3332 3332 3332 3332 3432 3332 3331 0 0 0 0 0 0 0 0 0  =7939
Rules:000001000=
S:80/1/A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1CC7/Q:1 #  T:1405 E:2 R:87/81/A=FEB FE2 FEB FE6 FE6 FE2 FEB 0 0 0 0 0 0 0 0 0  =35B8
S:80/3/B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4E88/Q:0 #  T:1410 E:2 R:87/83/B=3332 3332 3332 3332 3432 3332 3331 0 0 0 0 0 0 0 0 0  =5152
S:80/1/C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =647A/Q:1 # Rules:000001000=
 T:1404 E:2 R:87/81/C=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =D5CE
S:80/3/D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3635/Q:0 #  T:1410 E:2 R:87/83/D=3332 3332 3432 3432 3432 3332 3332 0 0 0 0 0 0 0 0 0  =D525
S:80/1/E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4C11/Q:1 #  T:1404 E:2 R:87/81/E=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =FDA5
S:80/3/F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1E5E/Q:0 # Rules:000001000=
 T:1410 E:2 R:87/83/F=3332 3332 3432 3432 3432 3332 3332 0 0 0 0 0 0 0 0 0  =FD4E
S:80/1/10=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C679/Q:1 #  T:1404 E:2 R:87/81/10=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =77CD
S:80/3/11=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9436/Q:0 #  T:1410 E:2 R:87/83/11=3332 3332 3432 3432 3432 3332 3332 0 0 0 0 0 0 0 0 0  =7726
Rules:000001000=
S:80/1/12=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EE12/Q:1 #  T:1404 E:2 R:87/81/12=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =5FA6
S:80/3/13=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BC5D/Q:0 #  T:1410 E:2 R:87/83/13=3332 3332 3432 3432 3432 3332 3332 0 0 0 0 0 0 0 0 0  =5F4D
Rules:000001000=
S:80/1/14=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =96AF/Q:1 #  T:1405 E:2 R:87/81/14=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =271B
S:80/3/15=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C4E0/Q:0 #  T:1410 E:2 R:87/83/15=3332 3332 3432 3432 3432 3332 3332 0 0 0 0 0 0 0 0 0  =27F0
S:80/1/16=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BEC4/Q:1 # Rules:000001000=
 T:1404 E:2 R:87/81/16=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =F70
S:80/3/17=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EC8B/Q:0 #  T:1410 E:2 R:87/83/17=3332 3432 3432 3432 3432 3332 3332 0 0 0 0 0 0 0 0 0  =FD6B
`
There are two thing, which confuses me:
- after installing the ESPControllerCode, only the first bank (bank0) is blinking through. The bank1 is not flashing...
- is it ok, if the ESP is not connected to the wifi (and the wifi-boot-webpage is shown), after flashing the esp?

Becouse the result is strange, here is the serial output of the normal ESPControler (from the inital issue)
`Found pcf8574
Connecting to WIFI
Connecting to Wi-Fi...
mode : sta(84:f3:eb:e4:6f:03)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with LB2, channel 6
dhcp client start...
S:80/1/1=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925/Q:5 # ip:192.168.8.101,mask:255.255.255.0,gw:192.168.8.1
Connected to Wi-Fi.
3. Connected IP:192.168.8.101
[... 150 lines cut ...]
S:80/1/49=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F54C/Q:15 # R:87/81/49=FE6 FE2 FEB FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =44F8

S:80/3/4A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =8F68/Q:14 # R:87/83/4A=3432 3432 3432 3432 3432 3432 3432 0 0 0 0 0 0 0 0 0  =37AB
Rules:000000011=
S:80/1/4B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =DD27/Q:15 # R:87/81/4B=FE6 FE2 FE6 FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =EED6

S:80/3/4C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =F7D5/Q:14 # R:87/83/4C=3432 3432 3432 3432 3432 3432 3432 0 0 0 0 0 0 0 0 0  =4F16
S:90/1/4D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =11C8/Q:13 # R:97/81/4D=FEB FD9 FE2 FEB FEB FEB FD0 0 0 0 0 0 0 0 0 0  =F994

S:80/1/4E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =11D4/Q:15 # Rules:000000011=
R:87/81/4E=FE6 FE2 FE6 FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =2225

S:80/3/4F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =439B/Q:14 # R:87/83/4F=3432 3432 3432 3432 3432 3432 3432 0 0 0 0 0 0 0 0 0  =FB58
S:90/1/50=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =2FEE/Q:13 # R:97/81/50=FEB FD9 FE2 FEB FEB FEB FCC 0 0 0 0 0 0 0 0 0  =C794

S:80/1/51=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =799/Q:15 # Rules:000000011=
R:87/81/51=FE6 FE2 FE6 FE6 FE6 FE2 FE6 0 0 0 0 0 0 0 0 0  =3468

S:80/3/52=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7DBD/Q:14 # `

I'm a litte confused... maybe you can help me?

> It has helped when timer for generating messages was increased to 12s - for my setup.

Ah, thanks, at which line/file can i change the timing?

stuartpittaway, 2020-02-16:
**There are two things, which confuse me:**

**after installing the ESPControllerCode, only the first bank (bank0) is blinking through. The bank1 is not flashing...**

The controller settings have been reset when you programmed this version in - I suspect the number of banks is set to "1" in the web page.

**is it ok, if the ESP is not connected to the wifi (and the wifi-boot-webpage is shown), after flashing the esp?**

Yes - the latest version formats the ESP8266 controller SPIFFs storage space and that also wipes the settings on the first upload.  Just go through the initial configuration again and all is good.

stuartpittaway, 2020-02-16:
Looking at the new trace, there are NO errors being reported by the controller.

You should see the controller requesting data S:80 and then S:90 - these are bank 0 and bank 1.

in your trace there is no S:90 so I assume that the setting isn't switched on for bank 1

stuartpittaway, 2020-02-24:
Hello, the latest code has drastically changed the way data packets are sent and received, so I'm closing this issue - feel free to re-raise if the new code doesn't work for you.

closed 2020-02-24 by stuartpittaway

Klauzen, 2020-03-17:
Thanks Stuart! It is working now perfectly with the new code!