stuartpittaway/diyBMSv4 #29: What is stability of design, want to place large order of assembled boards

kroesche, 2020-02-21:
Hi Stuart. I have a hobby project and we want to use this board design. I ordered a small number from OSHPark that was based on master about a month ago. I just order parts from digikey and hand assembled 3 so I could test them. The only change on master I dont have is the opto resistor change to 220ohm. However I notice you have a branch with a lot of commits that looks optimized for ordering from jlcpcb, and judging from comments in the commit messages I think you placed an order at some point.

My question is, how stable is this design now? Is there a point (commit number) that in your opinion is stable enough and known working that I could use it to place an order? I would like to use as much of your design "off-the-shelf" as possible. I would like to avoid the need to make any changes using kicad if possible.

I understand you can make no guarantees etc. Just trying to get your opinion about the maturity or stability. For example if you think there is still some problem area being worked on and you want to warn me off. I can make another small board order and hand assemble some more but am really looking to doing a larger order of assembled boards at this point.

Thanks for sharing your work on this!

stuartpittaway, 2020-02-22:
Hi, the design is stable. I've some spare boards from that last batch. Send me a message on the open energy forum and I can see if we can get some posted to you.

DGrebe, 2020-02-22:
Hi Stuart,

I am Dirk Grebe, living near Frankfurt, Germany.
I have seen your Videos and the GitHub pages. Great Work!

I am interested in 20 BMS Boards + 3 ESP Controller Boards.

What is the price?
Is it version 4.0 or 4.2? What would you prefer?

Kind Regards,

Dirk

> Am 22.02.2020 um 09:42 schrieb Stuart Pittaway <notifications@github.com>:
> 
> open energy forum

SvenSFS, 2020-07-18:
Answer in German:

Ich hab 5 Controller und 25 BMS Platinen v4.21 Bestellt. Die Kosten bei JLCPcb komplett 72€.
Also mit SMD Bestückung. Man muss NUR noch die Stecker und den ATiny aufbrutzeln.

Wobei 25€ nur Frachtkosten sind. Hab heute die Benachrichtigung bekommen das alles auf dem Weg ist.

Gruss 
Sven

closed 2021-05-17 by stuartpittaway