stuartpittaway/diyBMSv4 #31: Low temperature rule

Dolpin, 2020-02-24:
Brilliant BMS system and I am definitely going to build and use the v4 for an upcoming DIY electric boat conversion to a 16s 300Ah LIFEPO4 powered system.

I am only missing a rule to switch off the (solar) charger when a cell or pack temperature is below 0 celcius (<5 celcius is safer) as a lithium battery will be permanently damaged when charged in low temperature state.

stuartpittaway, 2020-02-24:
Thanks Dolpin, I'll try and get this new rule worked in for you.

What sort of boat are you powering ?

Dolpin, 2020-02-24:
Thanks Stuart, it will be a great addition to the BMS. I am designing a system for my Jeanneau Sun Light 30 sailing yacht using a Waterworld 7.5kW inboard

stuartpittaway, 2020-02-24:
Very nice

stuartpittaway, 2020-02-24:
Just a thought, at present the controller wants to connect to a wifi hotspot - it doesn't need the internet to run though - do you have a hotspot on the boat for it to connect with?

Dolpin, 2020-02-24:
I do not have a hotspot on the boat, I assumed I could use you my phone to setup and occasionally monitor the diyBMS controller. Is a internet connection required to run the system?

stuartpittaway, 2020-02-24:
No the internet isn't required to run but at the moment the controller only operates in "access point" mode during initial setup.  I'm sure its something I can fix at the same time.

Dolpin, 2020-02-24:
No problem as a hotspot is easy to use on a phone or tablet

stuartpittaway, 2020-02-26:
I've now implemented the "under" temperature rule - its in the master branch
https://github.com/stuartpittaway/diyBMSv4Code/issues/5

closed 2020-02-26 by stuartpittaway