stuartpittaway/diyBMSv4 #39: voltage referance

jency001, 2020-06-08:
hello,

Its not realy a problem more a suggestion.
the voltage divider u use to measure the voltage of the cell can this be changed by (example REF5040AIDR )? the 2 resistor you use to measure the voltage will change when the temperature changes. and this will be the case when you put the pcb on the battery. this chip would be a much more stable reference ( REF5040AIDR ) and will not change when the temperature changes.

pleas tell me what you think of this.

Kind regard

stuartpittaway, 2020-06-08:
Hello, I don't use this part at all in the 4.21 design - look at the JLCPCB branch for the BOM and circuit.

jency001, 2020-06-08:
hey stuart,

I know you dont use this part. 
i ment this would be an improvement on your design.
if you replace the voltage diveder(see picture) with this chip(REF5040AIDR) that would be an improvment. 
[image]

a voltage divider with resistors wil change when the temperature changes. 
this chip will not.

sorry i didn't explain it wright the first time.

Kind regards

stuartpittaway, 2020-06-08:
How would the REF5040 provide the voltage of the cells I'm trying to measure the voltage of?

The resistor divider measures the voltage of the cell its connected to.

jency001, 2020-06-08:
this chip is the same as your voltage divider but much more stable. and its not affected by temperature. And to answer your question Yes this would provide the cell voltage.

jency001, 2020-06-08:
this chip i told you is only one example i looked up there are other many better choices. i just wanted to inform you. i am thinking to change this building on your design

stuartpittaway, 2020-06-08:
How would it provide the cell voltage?

jency001, 2020-06-08:
sorry my bad.
i thought u used this divider to measure the cell voltage and then put this value in the micro controller to decide if its charged or not. is there a way i can talk to you ?

Kind Regard

stuartpittaway, 2020-06-08:
Correct, that is what the divider does, it is supplied (ENABLE line) with the cell voltage and divides it down to the ADC input on the ATTINY.

The 2.048V reference is seperately fed into the ADC reference pin

closed 2020-06-11 by stuartpittaway