stuartpittaway/diyBMSv4 #41: Add Russian language

sdqf-20, 2020-06-11:
Hello from Russia! I ordered circuit boards to repeat the project. I can translate the web interface and instructions into Russian, but I don’t see where the language data is stored. Sorry for my English if there are any errors.

Best Regards 
Evgeny

stuartpittaway, 2020-06-11:
Good idea Evgeny.

I would need to strip out the text into separate files before this would be possible.  Good suggestion.

sdqf-20, 2020-06-11:
Waiting file ;)

martinbogo, 2021-02-06:
Stuart,

I didn't want to open a new issue for this -- but I am happy to do translations to Spanish and Japanese.  Do you want help in refactoring the code to allow compile-time language substitution, or do you want language to be selected at configuration time post-flashing?  How much code space is there left in the ESP after flashing the current version into the D1 Mini Pro?

stuartpittaway, 2021-02-06:
Hi Martin, strange I started acting multi language support only a few days ago!

Have a look at the esp32controller branch for the code. I started with Spanish, but as I don't speak it, used Google translate, so expect it's poor quality!

martinbogo, 2021-02-06:
On it.  Checked out the branch just now... will PR Spanish shortly.