stuartpittaway/diyBMSv4 #42: Alternative Micro Controller just got compatible with Arduino IDE

happytm, 2020-06-23:
Hi @stuartpittaway ,

Thank you for you great work for DIY powerwall community. I wanted to see if attiny can be replaced with this low cost micro which has built in USB.

While searching for suitable micro controller for my other project I came across following repo. It makes this low cost micro controller chip arduino compatible. I think it definitely looks better than Attinys.  The pros are it has built in USB and it has up to 6 capacitative touch button inputs. Repo is brand new but it claims to make programming this chip easy. Repo is linked below:

https://github.com/DeqingSun/ch55xduino

for hardware design I found this:

https://hackaday.io/project/169671-ch552-dragon

Around 30 cents each and available at LCSC.

https://lcsc.com/search?q=ch552

Following  video  shows small footprint hardware with 2 touch buttons:

https://www.youtube.com/watch?v=EMDJe7JQ76g

What do you think? Please discuss.

Thanks.

happytm, 2020-06-23:
Following link have more information for different chips available from same company:

https://www.electrodragon.com/w/CH55X_HDK

Thanks.

stuartpittaway, 2020-06-24:
Looks a great little chip, I've no plans to replace the ATTINY at the moment - that is cheap enough to purchase and has a lot of software support.

closed 2020-06-24 by stuartpittaway

happytm, 2020-06-24:
Thank you for your feedback.