stuartpittaway/diyBMSv4 #43: Please add polarity markings to silkscreen

stienman, 2020-07-09:
After placing an order with jlcpcb, I received a request to confirm polarity of diodes and other parts. I checked the schematic and confirmed this with them, but it would be nice if the silkscreen included polarity, particularly for two pin polarized parts such as D2, D3, and D4.

“ Well got your order with many thanks~

Sorry to bother you, but there is one thing that we want to confirm with you about your SMT order before proceeding.

As shown below, we are not so sure about the polarity for the components below is correct or not, could you please kindly have a check?

Can we just make it as shown in the picture?”

[image]

stuartpittaway, 2020-07-09:
Thanks, this has now been completed and new files available in github branch.

closed 2020-07-09 by stuartpittaway

stienman, 2020-07-09:
Thanks for doing this, and so quickly!  I really appreciate your work on this project.