stuartpittaway/diyBMSv4 #44: jlcpcb DFM Check found error on Controller PCB

SvenSFS, 2020-07-12:
Hi 
i have order the Controller and the BMS V4.21 on JLCPCB.

In production Process fount the DMF an Error on U3 -> Soldering area is too small
Thanks for the great project.

[image]

stuartpittaway, 2020-07-13:
Hello, thanks for raising this issue.  We've had several hundred boards produced using the same GERBER files at JLCPCB !

The footprint looks fine to my eyes.  Perhaps delete your order and try again!

SvenSFS, 2020-07-13:
The job can be cancelled. it is in producion. 
i will test that.

I don't see any difference in the picture. but jlcpcb corrected that.

did someone else report this error?

atanisoft, 2020-07-13:
the footprint is set to SO-16-W and should be SOIC-16W, they are very similar but slight differences in the footprint (7.0mm vs 7.5mm on length)

stuartpittaway, 2020-07-13:
Nobody else has had the same problems!

Good spot on the solder pad sizes, I'll leave this issue open. As a reminder

stuartpittaway, 2020-07-24:
This is now fixed in the JLCPCBASSEMBLY branch.  Thanks for reporting.

closed 2020-07-24 by stuartpittaway