stuartpittaway/diyBMSv4 #45: Question: Different ATtiny841-SSU between ATTINY84A-SSU

SvenSFS, 2020-08-02:
is it possible to take the ATTINY84A-SSU?
the ATtiny841-SSU is hard to come by.

According to the document, the pin is compatible and has the same functions.
The price is the same in Germany

[Atmel-8495-8-bit-AVR-Microcontrollers-ATtiny441-AT-1315526.pdf](https://github.com/stuartpittaway/diyBMSv4/files/5011795/Atmel-8495-8-bit-AVR-Microcontrollers-ATtiny441-AT-1315526.pdf)
[ATTINY24A-44A-84A.PDF](https://github.com/stuartpittaway/diyBMSv4/files/5011796/ATTINY24A-44A-84A.PDF)

stuartpittaway, 2020-08-02:
No has to be the correct chip, they are different internally

closed 2020-08-06 by stuartpittaway

Jestergnet, 2021-01-30:
sorry, but the translator has not translated it unambiguously. Is this question also interested, yes or no?

Jestergnet, 2021-01-30:
> 
> 
> No has to be the correct chip, they are different internally

sorry, but the translator has not translated it unambiguously. Is this question also interested, yes or no?

stuartpittaway, 2021-01-30:
Diybms only works with the ATtiny841-SSU

Jestergnet, 2021-01-30:
> 
> 
> Diybms only works with the ATtiny841-SSU

Now understand. Thanks! It's a pity, the nonsense ordered ATTiny84a.