stuartpittaway/diyBMSv4 #46: ATTINY programming

jency001, 2020-08-09:
HI Stuart,

I am building your DIY bms but got stuck on the programming of the attiny.
i programmed the wemos alread that worked great. but for the cellmoduel i do not have an usb isp programmer.
i tried using an arduino and already look at several place on how to doe this but i did not have any luck.
Do you have an instruction on how to program the cellmodule with an arduino or someone that explains this very clearly.

Kind Regards

stuartpittaway, 2020-08-09:
Does this guide help?

https://www.microcontrollertips.com/programming-avr-microcontrollers-arduinos-arduino-isp/

You would use a spare Arduino Uno as a programming device.

jency001, 2020-08-09:
i already turned my arduino in an isp programmer with the steps you mentiond but its the next step i am stuck with.
so no i tought i need to open visual studio code again and just upload the attiny v400 to it but i fails.
 
[image]

stuartpittaway, 2020-08-09:
It's still trying to use usbasp programmer. Take a look at platformio.ini file and change usbasp to arduinoisp

jency001, 2020-08-09:
oke will try that . ill let you know if it works

jency001, 2020-08-09:
oke this got me one step further. i will try again tomorrow.
[image]
[image]

stuartpittaway, 2020-08-09:
You probably just need to tell platformio what com port to use.

jency001, 2020-08-09:
where do i fill this in i cant find it in the code.

stuartpittaway, 2020-08-09:
This should help...

https://docs.platformio.org/en/latest/projectconf/section_env_upload.html

jency001, 2020-08-09:
I tried changing the comport instead. Still same result. I will try to add the code tomorrow.
But I looked up the error code. It's something about the bootloader. Do I need to burn a bootloader first trough the arduino ide ?

stuartpittaway, 2020-08-09:
No the modules don't have a boot loader.

Are you using the upload using programmer option in platformio instead of the upload option?

jency001, 2020-08-09:
I do it the same way like for the wmos. I watched your video on how to upload the code. Open the program. Make workbook or something like that can't remember the correct term right know. The click on the alien head go to the v400 and click upload.

stuartpittaway, 2020-08-09:
If you are using Arduino programmer try upload using programmer instead of upload.

jency001, 2020-08-10:
No still no succès. I will try it tomorrow with an USB asp.

jency001, 2020-08-11:
hi Stuart,

so i tried again with an ATMEL-ice programmer. but still no success in visual studio. but i took the hex file generated from visual studio and uploaded it with Atmel studio and this worked fine. but i still have no idea whats wrong wit visual studio.
[image]

karciok, 2020-09-04:
Hi,
I think, that I figured out the problem.
In platformio.ini you need to add those 4 lines under upload flags

  -P$UPLOAD_PORT
  -b$UPLOAD_SPEED
upload_port = COM4   ;change COM4 to whatever port your arduino shows up as
upload_speed = 19200

and change 

upload_protocol = stk500v1

it worked for me, hope it helps anyone.

closed 2020-11-12 by stuartpittaway