stuartpittaway/diyBMSv4 #49: How much charging current can the BMS handle?

tusch001, 2020-09-18:
Hi Stuart,

thanks for all the work.

I know this project started with 18650 cells but Adam Welsh posted some weeks ago a video where he uses Calb battery cells. I was looking to build a bigger system and there are cells sold that can handle (up to and above) 176 A charging current.
  
I'm wondering how much current those resistors on the board can handle reliably over a long period of time in your experience. I think it doesn't make sense to buy big cells and then be limited by the bms. I couldn' find any info.

And if you don't mind me asking:
Is there a low temperature charging cut off in the sofware?

Thanks

stuartpittaway, 2020-09-18:
Hello,

First the easy question - yes low temperature cut off rule is available, you can use that to drive a relay to "do something".

The BMS doesn't directly control the charging or loan current into or out of the cells.  Instead it can drive a contactor or large relay to power connected devices on and off.

Each module connected to the cells can dissipate upto 0.85A, this is largely limited by heat and the surrounding temperature.

This may not be enough if you are looking at over 176A of charging current.  What are you using to power such a high charge current?

tusch001, 2020-09-18:
Hi Stuart,

thanks for the fast reply.

My project is in the planning phase.

I am working out of a car and have variable energy needs (e.g. in winter I would like to heat a cupboard 24 h to keep temps above freezing point). I also have limited charging time and limited access to an AC plug. I thought about a second alternator (if at all possible) or a dc to dc charger from the car (sorry - I don't know the English word) plus an AC battery charger. Not yet sure how much I can charge the batteries while driving. That's why I was looking at batteries that can be charged fast.

I've seen (12 V) battery chargers that deliver 30 - 80 A, the 176 A are the maximum specs of the above mentioned calb battery cells. Although I'm planning on 1p8s or 2p8s depending on the cell size.

I thought to choose the charger and battery cell size according to how much the bms can handle and calculate roughly if I can get anywhere near my energy requirements over 24 h without putting too much weight (and money) in the car.

I hope my answer makes sense.

Thanks for your help.

closed 2020-11-12 by stuartpittaway