stuartpittaway/diyBMSv4 #53: Controller: add CT clamp support

neographikal, 2020-11-21:
That way the contactor can be closed when the amperage going through the battery exceeds a set maximum for extra security.

neographikal, 2020-11-23:
> I'm also building an external current shunt to provide the DC current and voltage monitoring, which will connect back to the controller over RS485.

@stuartpittaway That is also nice to have. CT-clamp is very simple though, low cost solution. Would it be an idea to have that as an option as well?

stuartpittaway, 2020-11-24:
Hi @neographikal , aren't CT clamps designed for AC use?

neographikal, 2020-11-24:
I have none in my home at this moment, but there are DC-capable solutions. Fluke uses them in their meters:

https://nl.rs-online.com/web/p/clamp-meters/7657035/

thucar, 2021-01-29:
I've stumbled upon a few different products to try out for this.

- https://www.maglab.ch/products/cable-clamp/ - no idea of the cost, up to 230 amps, powered by +5V and returns an analog output from 500 to 4500mV
- https://www.sonnecy-shop.com/en/bidirectional-hall-effect-dc-current-sensor-cyhct-c2tc-output-4-20-madc-power-supply-12vdc-window-oe21mm-connector-phoenix.html - about 45€ plus shipping, up to 600 amps, powered from +12V and returns 4-20 mADC 
- https://www.evelta.com/wcs1500-hall-effect-based-linear-current-sensor-0-200a/ - about 3.5€, powered by 3-12V, returns analog voltage, linear to the passing current. (from 0.3V to VDD - 0.3V)

I'm getting ready to order a few of those to run some tests (I'd love for the DiyBMS to be able to track the capacity of my powerwall). 

@stuartpittaway - does any of the above sensors seem easier to implement than the others?

stuartpittaway, 2021-01-29:
I've not looked at using clamp meters, I generally find that they give a good approximation of current but not reliable enough for accurate monitoring.

Did you see the video I put together recently showing the new current shunt I've been working on?

https://youtu.be/DvLoORz5ff8

thucar, 2021-01-29:
I had not seen that video, thanks! I can now keep focusing on my powerwall installation and not worry about current monitoring solutions. It will mean that I'm going to need to switch over to that new controller to use that shunt though, correct?

stuartpittaway, 2021-03-19:
I'm closing this issue, as the current monitoring/shunt solution will cover this aspect of the monitoring system.

closed 2021-03-19 by stuartpittaway