stuartpittaway/diyBMSv4 #57: 50s configuration possible?

kaeferfreund, 2020-12-26:
Hi and many thanks for the awesome work everybody, especially stuart!

I am planning to cascade 50 DIY BMS modules. As far as I understand that is possible? The whole pack will be turned on an off via an automotive contractor which is able to handle up to 1000V and a lot more amps then I will ever use...

How many modules can be cascaded? Anything I have to look out for?

stuartpittaway, 2020-12-27:
Hello, the newest (unreleased) version of the code supports up to 100 modules on the ESP8266.

You have to be very careful once you exceed 40-50V DC !

Its here if you want to try it (experimental!)
https://github.com/stuartpittaway/diyBMSv4Code/tree/modulepwmchanges

closed 2020-12-27 by stuartpittaway

kaeferfreund, 2020-12-27:
Thanks for the quick reply.

I will do my tests on a fireproof surface with electrically isolated gloves (certified up to 1000V DC).

DC VOLTAGE IS REALLY DANGEROUS!

Thanks for the beta code. The current hardware revision of the BMS is capable of up to 100 monitors? No limitations etc?

stuartpittaway, 2020-12-27:
There shouldn't be any limitations up to 100 monitors/modules, as each module is isolated from one and another, the maximum voltage isn't really a problem.

Not many people have used this many modules (think 32 is the highest I've seen) so its probably untested configuration at 50, but should work, and I'm happy to help.