stuartpittaway/diyBMSv4 #59: comms errors

jdevine82, 2021-01-16:
Hi all
Just wanted to post my problem and solution for others in case they run into it. I had been experiencing comms errors between the controller circuit and the cell modules on and off for some time. I am using around 2m of cat5 running the tx and rx on the blue and green pairs. I tried adding better grounding, more filter capacitance and other things. However I found adding a diode on the rx pin of the pro mini (d7) to the 3.3v line to clamp the serial data voltage coming back to fix the problem. I just used a 1n4007.

Maybe on a future version of this circuit this might be a addition to the hardware.
Hope this may help someone.
Thanks very much to Stuart for his design. I appreciate it.
Jason

stuartpittaway, 2021-01-16:
Thanks for the fix Jason, D7 is currently pulled low by R2, what issues were you seeing with the data?  Did the module comms work fine, but the controller didn't receive the data?

I've recently had a conversation about this problem on the forum, in the end dropping R2 from 4k7 to 2k2 solved the problem.  If you get chance, perhaps try that instead of the diode/as well as.

It looks like the ESP is more sensitive to serial voltages than the ATTINY is.

jdevine82, 2021-01-16:
I had a look on the oscilloscope and it looked like the voltage at times seemed to be peaking higher than 3.3v. After it did this the esp seemed to lockout the pin for a time, before the waveform would go back to normal. I don't know what sort of internal protections the esp has. That is what gave me the the hint to try the diode clamping. I suspect it may be inductance in the cabling although pretty small current is involved in the comms, larger currents are always around from the inverter. I can see that dropping the resistor value should also fix the problem. I will try it sometime. The diode was easy as i could just jam it into the back of the pro mini header socket without taking it off the wall. The comms from the controller to the modules and between the modules has always been rock solid.

stuartpittaway, 2021-01-16:
Thanks for the detail, which way around did you put the diode ?

jdevine82, 2021-01-16:
Cathode to 3.3v.

stuartpittaway, 2021-04-12:
Further discussion on this topic is on going at the forum

https://community.openenergymonitor.org/t/workaround-for-the-communication-problems-change-resistor-in-controller/16388/70

closed 2023-01-03 by stuartpittaway