stuartpittaway/diyBMSv4 #60: PCF8754 addresses

stienman, 2021-01-18:
The PCF8754 comes in two versions for the footprint used in 4.21:
PCF8754T
PCF8754AT

This is so that you can have 16 on one bus, since each can be set to 8 different addresses.

The note in the source code is incorrect, it's not that there's a newer version, just that there are two versions, and the second is designated by "A" not "P".

I've changed my source as follows:
ESPController:main.cpp:90
//PCF8574A has an i2c address of 0x38 instead of the normal 0x20
//PCF857x pcf8574(0x38, &Wire);
PCF857x pcf8574(0x20, &Wire);

Since either chip can be used in any given installation, it might be worthwhile making it so they can both be supported, either through a configuration parameter, or automatically (test if 0x20 responds, if not, test if 0x38 responds).

FWIW, JLCPCB installed the PCF8574T (address 0x20) version on my boards with the default BOM provided in the project.

P indicates DIP16 package
T indicates SO16 package
TS indicates SSOP20 package

Sorry I didn't submit this as a pull request.

stuartpittaway, 2021-01-18:
I've fixed this in the next branch of code, support for both addresses

stienman, 2021-01-18:
That was quick, thanks!

closed 2021-01-18 by stienman

stuartpittaway, 2021-01-19:
The changes are in this branch.  https://github.com/stuartpittaway/diyBMSv4Code/tree/modulepwmchanges

I'm expecting to merge this into master later this week.