stuartpittaway/diyBMSv4 #63: V.4.20 cell monitors showing low voltage

thucar, 2021-01-27:
I’ve had a bunch of the 4.20 boards from JLCPCB sitting around for over half a year now. Finally decided the tested cells have been sitting long enough to weed out the self dischargers.

I’ve soldered Attiny’s on 7 on them and flashed with corresponding firmware. Strangely enough 4 out of those 7 cell monitors show 1.33V when hooked to a 100P 18650 pack. I have verified that the actual voltage is 4.15V and that is also th voltage I can see on the vcc and gnd pins of the programming header.

Moving the cell monitors from pack to pack confirms that the issue is with the modules thmselves. 

Will solder up and flash a new batch tomorrow. See if the results change.

Any idea what could cause such a behavior? I’m suspecting a voltage divider issue. What is the voltage range that should reach the attiny’s analog in pin doing voltage measurement?

thucar, 2021-01-28:
Bloody hell! I found the culprit.

17 out of the 30 boards I ordered, are missing the TL432CSF (D1) shunt regulator.
So mystery solved.

closed 2021-01-28 by thucar

stuartpittaway, 2021-01-28:
That will definitely do it!!

There have been a few problems with the quality of those TL chips.