stuartpittaway/diyBMSv4 #69: module v440 pcb issues

Thebradhome, 2021-03-04:
hello, i notice that the module v4.40 has some wrong cutout board issue in the design (see attached), also in the gerber files show the same problem, that cause the complete unconnection of the connector lanes to the board, i fix it and ive added an more professional design including all the component label into the copper layer, making it visible only looking the board from side, for a cleaner look (your boards are becaming a professional product so i think it wuold be nice to make it look professional as well) . i have new gerber files verified and obviously i'm glad to be contacted from you if i can help you in any way

issue finded: clearance, cutout, labels overwrite:

[image]
[image]
[image]

my idea of an possible solution:
[image]
[image]
[image]

stuartpittaway, 2021-03-04:
Thank you for the help and feedback.

I don't see the problems you highlighted on my designs or version of kicad, I've also used the Gerber's at jlcpcb without problems.

Are you running a new version of kicad?

LADesign-Github, 2021-03-04:
i'm using the newer one. 
Also i switched to eagle and easyeda because kicad is way limited in terms of font, and personalization (my point of view obviously). anyway, both of the software find me some error (maybe caused from importing files, i don't know), so i've decided to disturb you, showing the issue i found, hoping it could be useful.

stuartpittaway, 2021-03-04:
Ok, it looks like an issue with EasyEDA and new version of KiCad, so the problem doesn't exist on the existing design using KiCad 5.1.5

I'll upgrade to KiCad 6 in the future when its officially supported/released.

If you are importing the KiCad files into other design tools (Eagle/EasyEDA) then you will find issues as neither import the files 100% correctly.

Moving the silk screen labels to the copper layers looks nice, but also reduces the size of the copper ground plane

LADesign-Github, 2021-03-04:
yes, i'm with you that probably it's an importing issue.
talking about the reducing the copper ground plane moving the labels to the copper mask: yes it's true, but considering the amount of ground plane you have is absolutely irrelevant in that kind of circuits. you have all the ground plane you need without any kind of issues. i think it could cause some problem in some kind of circuit where you are on the limit of grounding plane and with some very sensitive device onboard . definitely not here. 
in my experience i've built a lot of circuit with that sistem and i never find difference in usability

stienman, 2021-03-04:
The copper ground plane isn't just for signal/noise purposes, but also
plays a role in heat dissipation.  In that case, and for this board
specifically, the rule is "The more the better!".

Personally I'm not a fan of having labels in the copper layer.  It's a
stylistic choice, sure, but as you move to more compact or complicated
designs there's simply no room.  While this may not fall into that category
I prefer to have the same style across all designs, regardless of
complexity.

-Adam

stuartpittaway, 2021-03-19:
I'm closing this issue, but thank you for the heads up on newer versions of KICAD may need some modifications to the existing boards to make them render correctly.

closed 2021-03-19 by stuartpittaway