stuartpittaway/diyBMSv4 #73: v4.4 module D2 rotate issue?

atanisoft, 2021-03-11:
Just checked the DFM output from JLCPCB for the v4.4 module PCBs and found the following differences:
Original:
[image]
Modified:
[image]

It appears that D2 is rotated 180deg.

stuartpittaway, 2021-03-11:
Hi Mike.

I think JLC have this WRONG - the "stripe" of the TVS diode needs to be facing towards the right handside of the board (towards positive).  

[image]

[image]

stuartpittaway, 2021-03-11:
I used this reference when building the circuit for the TVS... https://www.mouser.com/pdfdocs/AND8424-D.PDF

atanisoft, 2021-03-11:
> I think JLC have this WRONG - the "stripe" of the TVS diode needs to be facing towards the right handside of the board (towards positive).

Possibly, but usually they use "-" to indicate the stripe/marking on the diodes. If I receive them reversed I'll rotate it. But wanted to log this issue just in case it is in fact incorrect on the CPL file.

stuartpittaway, 2021-03-11:
This is the reference circuit.

[image]

and mine (note ground is on top, so flipped image)

[image]

I've just stolen this from https://www.quora.com/How-do-I-identify-a-TVS-diode

> A TVS diode will be connected directly across a supply rail (+5V or +12V), sometimes with a series inductor, fuse, polyswitch, or zero-ohm resistor. 
> **The anode of the diode goes to ground, while the cathode (striped end) connects to the positive supply rail.** 
> TVS diodes will usually be located near the power connector, but not always.
> 

I think JLC have this part recorded as a "standard" diode, so it looks backwards to them.

atanisoft, 2021-03-11:
Ok, as noted I'll check the boards when they arrive to confirm the orientation and adjust if required. 

We may need to get a note in the CPL/BOM file for JLCPCB to ensure proper rotation.

stuartpittaway, 2021-03-11:
Looks like JLC have told another user that the part placement is wrong as well.

Why do they care!!  I can understand if the placement was wrong or the foot print wrong!

stuartpittaway, 2021-03-11:
I'm starting to doubt myself on this now, I've checked that diagram so many times

stuartpittaway, 2021-03-11:
Think I've found the issue, I'm not rotating the part in KICAD however when the CPL/BOM files are generated by a script, its adding in a 180 rotation.

atanisoft, 2021-03-11:
Great, so as long as the part is correctly placed as the band to the + side (towards edge of board) it should work?

stuartpittaway, 2021-03-11:
Yes that's right. If it's the other way, it will be a dead short!

atanisoft, 2021-03-11:
Gotcha, I'll confirm before testing/flashing/etc. Hopefully I'll have the boards in a bit over a week. I'll also use six jumper wires to test flashing from the ESP32 controller.

stuartpittaway, 2021-03-16:
Forum posts have indicated that this image/placement

[image]

results in a board being made like this

[image]

**THIS IS INCORRECT, SO ENSURE THE PREVIEW IMAGE SHOWS NEGATIVE ON THE RIGHT HAND SIDE**

Paragon35, 2021-04-04:
Hi, Stuart I got  modules from JLCPCB where D2 has a cathode stripe (-) gray stripe is on the left and an anode (+) without stripe is on the right so is it ok?

[image]

stuartpittaway, 2021-04-04:
No that's incorrect.  Do not power up these boards before fixing the issue.

Two options

Unsolder and remove the part

Or

Unsolder and rotate, solder back on.

The part is quite large, so not difficult to do.

Paragon35, 2021-04-04:
> 
> 
> No that's incorrect. Do not power up these boards before fixing the issue.
> 
> Two options
> 
> Unsolder and remove the part
> 
> Or
> 
> Unsolder and rotate, solder back on.
> 
> The part is quite large, so not difficult to do.

thanks so when JLCPCB wanted to turn the element unnecessarily I told them to leave them as in the original :)
I will turn around, no problem, as you write it is a big element, thanks again and best regards

[image]

stuartpittaway, 2021-04-04:
How long ago did you order these? I changed the silkscreen a few weeks ago to help jlc orient the part correctly

Paragon35, 2021-04-04:
> 
> 
> How long ago did you order these? I changed the silkscreen a few weeks ago to help jlc orient the part correctly

Stuart, order from 2021-03-10

as you can see in the picture I pasted above, JLCPCB wanted to correct the cathode (-) on the right side, it's my fault because I insisted they leave as in the original where the cathode (-) is on the left side

closed 2021-05-17 by stuartpittaway

SP3S, 2023-04-12:
https://user-images.githubusercontent.com/62333994/231421323-ccdf93c0-a129-4373-8495-a3a33c1abfde.png