stuartpittaway/diyBMSv4 #76: JLCPCB doee not accept 4.40 pcb

fhorst1, 2021-03-20:
JLCBCB rejected the 4.40 PCB as they claim it have V-cut.

Something they don't seem to support.
[image]

As JLCPCB is crazy and direct repays the payment made with PayPal, you are screwed.

You can confirm and override their rejection.
But your money is already back.

And if you have the luck to not have USD bank account..
It will be deducted in euro, converted at bad rates to USD.
Then at the repay in USD, converted back to Euro with again really bad rates.
It will cost about 15% 
And you end up with nothing to show for...

Not directly related to the design, but detect result of the V-cut used.

At single boards it might be OK
When you panalise with stamp hole, this V-cut will be a problem.

Please change the V-cut for the temp sensor to a stamp hole.

Thank you

fhorst1, 2021-03-20:
Edit:
This happens when you panelise the boards.
I tried 4*2 to order sets of 8.

That didn't work out.

stuartpittaway, 2021-03-20:
I don't use vcuts, as jlcpcb don't support them.

If you have selected to panel the board, then the design will have changed and I can't control that, or the output of it.

closed 2021-03-20 by fhorst1