stuartpittaway/diyBMSv4 #77: DIYBMS Controller V 4.2

Paragon35, 2021-03-29:
hey Stuart, today the controller got to me and I have such doubts if you could spread it :) ..
Your controller in the selected two sections has soldered elements:

[image]

that are not on the JLCPCB controller if this is a problem, 

[image]

I checked the parts list before ordering and they were all, greetings from Poland Leszek ....

stuartpittaway, 2021-03-29:
What you have is correct, and the latest version - none of those parts are in use.  

I've removed the TX2 and RX2 port from future revisions of the board.

Paragon35, 2021-03-29:
> 
> 
> What you have is correct, and the latest version - none of those parts are in use.
> 
> I've removed the TX2 and RX2 port from future revisions of the board.

Thanks Stuart, the project is really impressive. Congratulations, I'm picking up the parts with parcels. Modules have not arrived yet, thanks again and best regards Leszek

closed 2021-03-30 by Paragon35