stuartpittaway/diyBMSv4 #79: Is ballancing done right ?

axwell, 2021-04-12:
Hello Stuart, i have a question about the balancing.
My 6s6p pack will charge within a voltage range of 10-20 mv between cells, but after the 4.1 threshold the pack is instantly debalanced some cells going sometines to 4.3

  from 4.1 volts for a short period of time and range to about 200 - 300 mv difference.

Why? is this ok ?

Also shouldn't balance start right away when charging, progressively starting to apply PWM to the cells that are "overcharged" ? This avoids high temperatures because you don't need to start with 100 % PWM or when the cell is almost charged ?

[image]

fhorst1, 2021-04-12:
I don't know about your type of cells.
4.3 seems high to me.
Lithium polymer has top of 4.2?
I'm more LiFePO4 guy (top 3.65)

You don't need to start at lower voltages.

If you do a battery capacity test, you will notice that the cells don't have the same voltages at similar SOC.
At 60% my testing voltages during 10A load varry between 3.1 and 3.3, where they all end at the same bottom voltage with the same capacity.

Charging will give the same issue, and you might be holding back a cell who is in reality lower on capacity, just shows little higher voltage.

I have the balance set at 3.5v, enough room to hold back, if needed.

For polymer, 4.05 is similar.
If the top is 4.2, no cell should be able to charge to 4.3!!!
Your controller board should stop the charge after one (or more) cells have reached the maximum voltage.

axwell, 2021-04-12:
batteries are panasonic ncr18650b, all new.
yes, it stops, but after the voltage spike of 4.3 for a few seconds.

stuartpittaway, 2021-04-12:
How are you charging them? Ideally the charger stops and balancing is then rarely required.

At present, it's a simple on/off top balance, it works but is a little crude.

axwell, 2021-04-13:
charger is victron mppt, set at 24.6 V.
It will not turn off, because the load will switch off also.

fhorst1, 2021-04-13:
I have a "hybrid inverter" so MPPT build in.

I can't switch off the Mppt without the inverter, but I can (and do) disconnect the solar power coming into the MPPT with a contactor.

Please don't use Solid State Relay for this, they get too hot and aren't intended for long contact periods, rather many short contacts for a long, long time 

Magnetic Contactors, 20-50A should be enough, available in 5 to ?? DC volt.
That means the magnetic coil is switched with that voltage.
The contact voltage can be 800-1000v, at 20-50A.
Most people use 12v contactors.

Some say they use a lot of energy.
My cheap Chinese use 5w keeping contact.
Up to you to decide if that is a lot or not :-)

[image]
This is a good quality one, uses less energy (about 3w) and is +$100,-

[image]
This is my cheap type, price about $20,-

fhorst1, 2021-04-13:
Your controller board can have the relay board.

[image]

Each really can disconnect the contactor at a rule.
In this situation, when the voltage is above 4.205v (or so) disconnect.

And automatic reconnect at your settings, like 4.15v or so

stuartpittaway, 2021-04-13:
> charger is victron mppt, set at 24.6 V.

On the screenshot of diybms it shows 24.95V - how did the voltage get that high if the Victron isn't set to go that high?

axwell, 2021-04-13:
> > charger is victron mppt, set at 24.6 V.
> 
> On the screenshot of diybms it shows 24.95V - how did the voltage get that high if the Victron isn't set to go that high?

Neither do i, that's why i was asking the question about the balance.
When one of the cell goes into balance, other 1-3 cells goes to 4.3 instantly and overall votage overshoots, then i have the relay board from above and the relay kicks in.

fhorst1, 2021-04-13:
At what voltage do you have your balancing set?

It seems like they are really close together, and almost reach the balance top of 4.2 at the same time.

Even if one cell does get discharged , it should not affect the other cells. (That much)

Only thing I can think of, is that the charger goes from CV, constant voltage charging stage (the least stage) back to CC, constant current.
Or atleast higher capacity.

Overall voltage does reduce, what might trigger the charger.

Usually the dumpload of the DIYBMS doesn't have that much impact, except on low capacity cells.
How many of the panasonic ncr18650b do you have in parallel??
If it's just a few Ah, that would explain a lot.
(With 560Ah, one DIYBMS cell module doesn't have much impact..
Even with 3 cell modules, it's just slowly balancing/dumping..)

If it's just one or two cells, the dumpload might be too strong for the capacity.

I suggest to play around a little with both the balance start voltage.
Perhaps let it start at 4.0v.
And slowly work your way up to the level where you feel comfortable that it doesn't swing so much.

 The charge controller settings:
It does seem to be 0.3v lower at the controller then the battery is, you could adjust the victron to 24.3v

Usually there are 2 settings, bulk and absorption phase.

I usually have my bulk phase one volt higher then the absorption.
(Absorption is also the CV phase)

I hope this helps.

axwell, 2021-04-13:
I lowered the voltage on victron.
My pack is 20Ah, maybe is too small for the balacing ?

fhorst1, 2021-04-14:
The standard 0.8A might be too much for your small pack.

It's "easy" to adjust. (@stuartpittaway , please correct me if I'm wrong about this)

While you can't really tune the capacity, it's hold back by temperature.
Currently set at 60 degrees Celsius.
With a heatsink the temperature is lower, making it possible to have higher dump load (up to 1.2A)

When you reduce the temperature to like 50C, it should dump less current.

fhorst1, 2021-04-18:
@axwell 

Any update if lowering the maximal dumpload temperature helped?

axwell, 2021-04-21:
@fhorst1 kind of works, when balancing starts until it reaches the temperature the voltage overshoots, but after temperature is reached is ok.
I think a better option is to limit the current, it was nice to be a setting in the interface.

stuartpittaway, 2021-04-21:
You can't limit the current, just control the PWM frequency using the temperature.

fhorst1, 2021-04-21:
Happy that the workaround did the trick.

20Ah is for this type of BMS "really small" 

In my experience battery is always too small.
For me that stopped after hitting the +1000Ah @s16 LiFePO4 (51.2v)

If you have a possibility to grow, it's working for now, and later will be better.

No possibility to grow? It's working for now ...... :-)
But DIYBMS might not be the best choice.

I do believe in obtainium, use what you have!

stuartpittaway, 2021-04-21:
You will probably struggle to find a BMS with a higher balance current, diybms and batrium seem to have the highest loads at the moment. Somewhere between 0.6A and 2A

fhorst1, 2021-04-22:
It seems like the high Balance capacity is the problem for this (small capacity) setup.

With 20Ah, this BMS might not be the best choice. It's functional now, that's most important.

And if you plan on expanding, it will be great.

I doubt there is anything else to fix this problem.

Perhaps @stuartpittaway can make in the introduction of DIYBMS a remark about adviced minimal capacity of 40Ah.
(The 40Ah is educated guessing, where 20Ah is doable with restricted temperature, double it should be OK)

axwell, 2021-06-30:
Hello again,
even with a lower temperature the issue was more or less the same, so i decided to build a bigger pack and everything was setup up the same issue appeared, so i started to wonder if the cables were ok.

I had the cable like this:
[image]

So i decided to make 1-1 cables and a little bit thicker and the problem went away, balancing is perfect now :)

SergioRius, 2021-08-12:
> At what voltage do you have your balancing set?
>
> It seems like they are really close together, and almost reach the balance top of 4.2 at the same time.
>
> Even if one cell does get discharged , it [...]

So interesting. I had my +200Ah bank configured at 4.05v (victron equipment) and i was having fast battery voltage raising over the charge limit when having big ac loads, like charging the car.
I've lowered the voltage limit to 3.93v/c to avoid blowing cells since.

If absorbtion is set to 55,3 and float to 55v, what do you think bypass should be set?

datenheim, 2022-08-10:
> I had the cable like this: [image]
> So i decided to make 1-1 cables and a little bit thicker and the problem went away, balancing is perfect now :)

I Also started with these far too tin cables and had a lot of issues with balancing. As soon as the balancing started to kick in, the measured voltage was modified by the voltage drop of the balancing current (balancer and voltage measurement share the same thin wires).

I would suggest to either use thicker wires, or even to separate the balancers current path from the measurement by using 4 wires - 2 for balance current, and another 2 for voltage sense...

closed 2023-07-05 by stuartpittaway