stuartpittaway/diyBMSv4 #81: Type of esp32 WROOM

fhorst1, 2021-04-22:
Issue: please provide detailed information about the correct ESP32 

There are sadly many different types, easy to buy the incorrect one.

It took me some digging via the Amazon link to Google and official site to find what type I really need.

If all is correct it should be the ESP32 WROOM 32D

**NOT the 32 or 32U, ONLY the 32D.**

I'm not sure if it needs the 4, 8 or 16 MB module.
I got stuck there.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards.html

stuartpittaway, 2021-04-22:
The 32U also works, that one has an external WiFi aerial.

The important part of the esp32 is the board it's attached to, which must be the devkitc variant, so check the pin out of the device before purchasing.

There are some amazon links in the GitHub pages to modules I've brought and used.

stuartpittaway, 2021-04-22:
4MB is used on the devkitc devices, and that's what the BMS code is designed for.

atanisoft, 2021-04-22:
Make sure you are getting the DUAL core variant of the ESP32. There is a DevKitM-1 variant that uses an alternative pinout and is also single core, this is *NOT* supported by arduino-esp32.

fhorst1, 2021-04-23:
Thank you both for the answer.
This issue can be closed.

As I mentioned, just Amazon links makes it a real search to hope to find the correct board.

The Amazon link doesn't provide enough detailed information needed to find it on different sources.

Mentioning the full name might help others to purchase the correct one.

stuartpittaway, 2021-04-23:
The problem with Amazon is that is appears to change the product link if the item is out of stock or not available in some regions.

Often it repoints to a similar but not identical product link.

closed 2021-05-17 by stuartpittaway