stuartpittaway/diyBMSv4 #82: V440 Module Bypass Resistors

pfischer54, 2021-05-17:
These resistors (R30,R28,R26,R24,R29,R27,R22,R13,R11,...) have been on "inventory shortage" for some time at JLCPCB - for me at least!  I noticed that one supplier shows these at "Not recommended for new designs"):  [LCSC C21624](https://lcsc.com/product-detail/Chip-Resistor-Surface-Mount_Uniroyal-Elec-1210W3F330KT5E_C21624.html) 
Maybe this is why JLCPCB are not re-stocking them?

There is an alternative which is slightly larger, higher wattage, and which JLCPCB stock: [LCSC C61431](https://lcsc.com/product-detail/Chip-Resistor-Surface-Mount_UNI-ROYAL-Uniroyal-Elec-1812W2J033JT4E_C61431.html)

so would require some moving parts around the pcb...

Is this a reasonable change?

stuartpittaway, 2021-05-17:
They were in stock when I looked this morning!

pfischer54, 2021-05-17:
I've been checking with an older set of files, that will teach me :) .  I can see the resistors have long since been changed...
Thanks Stuart

closed 2021-05-17 by pfischer54