stuartpittaway/diyBMSv4 #83: TJA1057GT/3J

trebor1508, 2021-05-17:
is a shortage of this element in jlc and everywhere. can be replace by other available chip?

atanisoft, 2021-05-17:
IFX1050GVIO should be a drop in replacement, but it is also likely backordered from most suppliers.

You can also omit this chip safely if you are not planning on using the CAN bus for taking to a charge controller etc.

trebor1508, 2021-05-18:
For what purpose this communication can be used? I have a victron equipment. 
Or i think i can solder manually later on when this chip will be available.

stuartpittaway, 2021-05-18:
It can be added later on without a problem - its not currently used by the software so you won't miss any functionality (at the moment).

I've spoken with JLC and they have said there should be some arriving this week - but I've been told that before.

stuartpittaway, 2021-05-18:
Parts which can be used:

IFX1050GXUMA1 = C129255
TJA1057GT/3J = C112947
TJA1051TK/3 (no longer available from JLCPCB)

On the TJA range - the import part is the "/3" - indicating its 3.3V compatible.

At the time of writing, JLC have very little stock of the above.

Looking at other sites, all are out of stock (Farnell/Mouser etc.)

trebor1508, 2021-05-18:
I have check for C112947 all available sources and nothing.

https://b2b.partcommunity.com/3d-cad-models/sw-push-6mm-h5mm-kicad?info=kicad%2Fbutton_switch_tht2e3dshapes%2Fsw_push_6mm_h5mm.prj&cwid=8655
this kind push button?

Yes I will be ordering now.  So i will try 4,3

about Farnell, part of group called CPC is located 1miles from my home location (Preston) :)

Thanks Stuart

stuartpittaway, 2021-05-18:
That button should be good. Just a word of warning, this is a new revision but nothing major has changed, but I've also not fully tested it!

trebor1508, 2021-05-18:
So it will be a test, I will order 5 fully asembled (difference in price is about 10F)

trebor1508, 2021-05-19:
Stuart. Jlc said "The orientation of the component does not match the pad" on controller board. But i do not see any changes.

fhorst1, 2021-05-20:
I'm a bit confused.
The new version controler can "talk" CANBUS? Or not yet?

For me the communication with canbus was / is the reason to switch over to the new controller version.

Due to problems with parallel, I now have my setup all series, and need 5* controller, 80x cell modules.
Not a problem, I have.

My 3 hybrid inverters have only CAN as communication protocol for the BMS.
Not sure how it can actively assist the charge controller, that is something to find out once installed :-)

Would be nice if it already can do CAN :-)

stuartpittaway, 2021-05-20:
The CAN hardware is on the controller board, but the software has not been written to talk CAN - yet.

What inverters do you have?

stuartpittaway, 2021-05-20:
> Stuart. Jlc said "The orientation of the component does not match the pad" on controller board.
> F1
> But i do not see any changes.

Thanks @trebor1508 - I changed the size of the pad for the fuse (F1) as JLC kept saying it was too large - the part cannot be oriented incorrectly as it doesn't have a polarity.

Can you upload the pictures they send you (or you can logon to the JLC account to see them)

fhorst1, 2021-05-20:
I found the communication protocol.

Can't seem to be able to attach PDF here, so I did it in OEM forum

trebor1508, 2021-05-20:
[image]
[image]

stuartpittaway, 2021-05-20:
> I hope this helps

It looks identical to me as well !

trebor1508, 2021-05-20:
They mention about resistor R3 but R1 and R4 looks identical. 
F1 should be good as well.

stuartpittaway, 2021-05-20:
@trebor1508 think I've spotted an issue with the controller board - U9 and U12 are rotated incorrectly, this will need to be fixed after delivery/build of your order - unless you can cancel now.

trebor1508, 2021-05-20:
I can not cancel now. But it is only 4 leg chip. I think i can fix it when they arrived.

trebor1508, 2021-05-20:
Stuart

> @trebor1508 think I've spotted an issue with the controller board - U9 and U12 are rotated incorrectly, this will need to be fixed after delivery/build of your order - unless you can cancel now.

check the photo what I upload, JLC rotate U9 and U12. So I do not have to do anything.

stuartpittaway, 2021-05-20:
Great, should be good to go then.

trebor1508, 2021-05-20:
> Great, should be good to go then.

I placed my order yesterday.

trebor1508, 2021-05-21:
one more simple question: Switch SW1 and SW2 how many milimeters are between legs?

stuartpittaway, 2021-05-22:
This is the part the footprint is based on

https://www.mouser.co.uk/ProductDetail/CK/PTS645SL502LFS?qs=UxeAxwACbqm88cEAxDT3kg==

It's fairly generic foot print, think it's 6mm spacing, but check datasheet

trebor1508, 2021-05-27:
Is here someone from UK who order from JLC recently?  I have choose DHL and they sent me a message to pay £45 import tax. But on JLC was on order over £135 will be no tax. 
£158 order with postage €45 tax. SWEET. Bloody BREXIT.

stuartpittaway, 2021-05-28:
When you order there is another. Option to use DHL with prepaid tax, either way you have to pay vat at 20%

The message on jlc website is incorrect

trebor1508, 2021-06-02:
I received my order from JLCPCB

I have solder the first controler and is working fine. 

[image]
SD CARD socket is not so good.

[image]
[image]

trebor1508, 2021-06-02:
[image]
Is this fine? Or cable color is reversed?

atanisoft, 2021-06-02:
Module cables are reversed. Use a thin flat blade to pull up the securing pin and swap them.

stuartpittaway, 2021-06-03:
What is wrong with the SD card holder? Does it work okay?

trebor1508, 2021-06-03:
It is working but screen need to be removed to install sd card.

stuartpittaway, 2021-06-03:
> It is working but screen need to be removed to install sd card.

That's a relief, I've ordered 40 boards!

The new card holder is more secure, and the card won't fall out due to vibration etc.

trebor1508, 2021-06-03:
To be clear,  i am not complaining. I am sharing my thoughts and observations.

stuartpittaway, 2021-06-03:
No problem, I didn't read it as a complaint, just making sure it all works!!

trebor1508, 2021-06-03:
Good. I will do some modules and seehow it works with them.

trebor1508, 2021-06-03:
[image]
So this plus belong to connector? I think i bought wrong cable.

atanisoft, 2021-06-03:
> So this plus belong to connector? I think i bought wrong cable.

Yes, the red/black need to be reversed in the cable. Unfortunately almost all cables are sold in this fashion. You can use a thin blade to pull up the locking pin on the cable to remove the wires and reinsert them in the correct orientation.

trebor1508, 2021-06-03:
what mean "libusb0.dll" can not be found? I try to program module.

shampeon, 2021-06-06:
> Parts which can be used:
>
> IFX1050GXUMA1 = C129255
> TJA1057GT/3J = C112947
> TJA1051TK/3 (no longer available from JLCPCB)
>
> On the TJA range - the import part is the "/3" - indicating its 3.3V [...]

Would this TI chip be a good replacement for the TJA1057GT/3J? Mouser has plenty in stock. 
https://www.mouser.com/ProductDetail/Texas-Instruments/TCAN4420DR?qs=W0yvOO0ixfFdDbLci2afMw%3D%3D

atanisoft, 2021-06-06:
@shampeon from a very quick review of the datasheet it looks like it might work. It does have a considerably higher cost basis though, around 3x of TJA1057GT/3J and 2x of IFX1050GXUMA1.

fhorst1, 2021-06-06:
At $2.72 total price will be 2 bucks more, per module.

For low cost, the older controller board works fine.

For more (future) advanced controller, the 2 bucks extra is probably not going to be the showstopper.

I found with my order Many parts unavailable from JLCPCB, and just ordered with what was available.
The rest via components shop like Rs-online.

stuartpittaway, 2021-06-06:
As the CAN chip is not used at the moment, I'd wait and add the correct part when available

trebor1508, 2021-06-08:
Up to what voltage cell modules can be safely connected?

phisp, 2021-06-08:
i tried 6V with the V4.00 and the leds got very bright but it worked. According to the Attiny Datasheet: 1.7-5.5v, however the Optocoupler and the Leds will fail probably first. Voltages below the referance Voltage will result in false measurements. And some boards started to balance the cells. Somehow you could program the Attiny to shutdown, if the voltage is getting to low, however i was not able to find it.

stuartpittaway, 2021-06-08:
> Up to what voltage cell modules can be safely connected?

The modules are designed for maximum of 4.35V, but most of the components work to 5V maximum

trebor1508, 2021-06-08:
> > Up to what voltage cell modules can be safely connected?
> 
> The modules are designed for maximum of 4.35V, but most of the components work to 5V maximum
> 
> The attiny is programmed to shutdown at 1.8V

Do you think about modules to work with leisure batteries in campers for example?

stuartpittaway, 2021-06-08:
Is there a need? Lead acid don't need balance, the new current monitor (out soon!) would measure voltage as well (up to 85v).

trebor1508, 2021-06-09:
When i can change numbers of modules settings before i connect them to controler with loop RX TX? Or modules should be connected and changing setting then? I see "failed to save settings"

stuartpittaway, 2021-06-09:
Just press F5 to refresh the webpage if you see the save error.

trebor1508, 2021-06-09:
[image]
Why this is happening?

stuartpittaway, 2021-06-09:
It can't talk to the attiny chip in the module, have you connected it correctly to the ISP header?

trebor1508, 2021-06-09:
Yes.

trebor1508, 2021-06-09:
[image]
And 1 module have incorrect voltage readings.

stuartpittaway, 2021-06-09:
Have you calibrated the modules?  Switch off balancing before calibration though

trebor1508, 2021-06-10:
> Have you calibrated the modules? Switch off balancing before calibration though

I am so sorry but i have watched all your youtube films and tried to find on forum how to do that but i have no idea.

stuartpittaway, 2021-06-10:
You need to calibrate each module against the cell using a good quality known accurate multimeter.

Measure the voltage of the cell (when idle/not balancing or charging) and enter it into the module page, configuration setting.

Until then, the cell voltages will vary.

trebor1508, 2021-06-10:
It works. Thanks.
And how to switch off balancing?

stuartpittaway, 2021-06-11:
Balancing stops when the voltage is lower than the balance voltage set in the web interface.

If the modules are going into bypass straight away, try and calibrate with a different cell at a lower voltage

SergioRius, 2021-08-12:
On to the TJA1057GT/3J topic, do something need to be done to the jumper pad near the top of the can chip, or is totally unrelated?
I have this pad without solder and I'm afraid is something necessary for the TJA1057GT/3J to work.

SergioRius, 2021-08-12:
> @SergioRius
> 
> > do something need to be done to the jumper pad near the top of the can chip, or is totally unrelated?
> 
> The jumper selects the CAN bus voltage (5v or 3v3) and defaults to 5v which is the most common CAN voltage. IF the hardware you are connecting to uses 3v3 CAN voltage you can cut the trace between the center pad and the 5v pad and then add a solder blob to connect the center pad to the 3v3 pad. But likely you can leave it as-is.

Thanks, i didn't appreciate a track between 5v pads.

closed 2022-04-14 by stuartpittaway