stuartpittaway/diyBMSv4 #85: implement CAN Interface

fhorst1, 2021-06-05:
Please add CAN support for the CAN port.

the request was also made here: https://community.openenergymonitor.org/t/diybms-v4/11292/3697
(with PDF to download)
The same information I have about the CAN usage for my inverters is inserted as jpg, as PDF can't be shared here :-)

-  CAN 2.0B
-  CAN Baud Rate：250kbps
-  11-bit identifiers
-  Little endian

[image]

[image]<br /><a target='_blank' href='https://imgbb.com/'>best image hosting</a><br />

[image]<br /><a target='_blank' href='https://imgbb.com/'>best image hosting</a><br />

[image]<br /><a target='_blank' href='https://imgbb.com/'>best image hosting</a><br />

stuartpittaway, 2021-06-05:
Are you looking for the diybms to be a slave to the charger/inverter?  Will it have to answer questions or will it be transmitting to them?

fhorst1, 2021-06-05:
Only transmit information.
So master :-)

BMS = Battery **_Management_** System.

We can't have two captains on one ship!!

It's a valid question, as CAN is 2 way protocol.

It could have been so simple.

Apparently the Chinese designer of the hybrid inverter / battery pack wanted to look extra smart by using only CAN...

Or actually, if I understood it correctly, they where already selling 5 kWh LiFePO4 battery packs, S15 and used that system as guideline for BMS/LiFePO4.... 
What just happened to be CAN 

Their standard LiFePO4 settings is for S15 only.. 
And where flabbergasted when I told them S16 is more or less industry standard.

Obtainium, without market research..
Can bite you in the @ss 

In my situation, the hybrid inverter has absolutely nothing to tell to the BMS, as I don't want the BMS to be configured via the inverter.
(We got the nice web interface)

What I hope will happen is that the MPPT charge will automatically be adjusted when it receives more accurate information about individual cell voltages, and not just pack voltage.

The voltage shown isn't accurate, it shows 52 when the cells are 53, but the difference is always the same, so easy to adjust the (manual) charging protocol to correct this voltage offset.

Perhaps in the future if the smart shunt is implemented, it can give more accurate information about battery usage.
That would be nice.

Thank you!!