stuartpittaway/diyBMSv4 #87: High number of modules - time out issue

emir0815, 2021-06-07:
I have 5 strings, each 14 cells liIon. I have cell module 4/1 (=module 57) blinking blue and green during the process of the controller gathering data from each module.

According to Stuart, this is a timeout issue and could be a bug. 
The whole roundtrip takes 13seconds.

My timeout issue has now changed to module 3/5 which is number 33 in the row.

regards
Karl

stuartpittaway, 2021-06-07:
Hi Karl, thanks for raising this.

What happens is this:

1. Controller sends a message to the first module
2. Module processes message and passes to next module (and flashes green LED at the same time)
3. Modules repeat above until the message gets back to the controller

The controller may send additional messages a few seconds after the first, so you may see a "chain" of green LEDs going down the modules.

The message can only communicate with 16 modules at a time, so multiple messages are sent to "talk" to modules over 16.

In addition to the above, each module is independent in that it will self monitor and balance a cell, even without talking to the controller (as a safety feature).  This check kicks in every 8 seconds, and it when the blue LED lights up.

This check only occurs if no message has been received from the controller (or previous module) during this time.

The ATTINY chip used has a maximum of 8 seconds delay between these checks (hardware limitation), so I will go away and check the code as you also mentioned that the configuration data is not being saved.

Can I ask, do you see any error messages in the web interface, lost packets etc?

emir0815, 2021-06-07:
On a total of 65k packets sent, I have 51k packets received, 131 OOS errors and 2 CRC Errors...

emir0815, 2021-06-12:
voila 
[image]

stuartpittaway, 2021-06-12:
It might be worth pressing the reset counters button, but from the screen shot it shows module 12 has something wrong with it. Packet counts a lot higher than other cells

emir0815, 2021-06-12:
Ok, how can this happen?

emir0815, 2021-06-12:
[image]

emir0815, 2021-06-14:
here are some more screenshots
[image]
[image]
[image]

stuartpittaway, 2021-06-14:
So the pictures tell me that there is something wrong with module 12. Up to that point there are no bad packets.

It also looks like the module has recently rebooted as the packet count is low.

emir0815, 2021-06-14:
Yeah, i did reboot because of low response time and you wished a counter reset 🤪

emir0815, 2021-06-14:
Shall i replace module 12?

stuartpittaway, 2021-06-14:
> Shall i replace module 12?

You could just swap it with another module if you have a spare.

All the packet counts should be the same you see.

emir0815, 2021-06-15:
Well, e.g. module 12 sometimes shows bullshit voltages. today it showed 3.2V where cell had 3.9V. When I calibrated it, it sprang up to 4.3V. Then I had to re-calibrate. And it is not the only module behaving like that.
There is definitely a communication issue...

I'm going to swap out these modules that show lower count on received packages... lets see if that changes something.

thx stuart!

phisp, 2021-09-15:
I had also an issue with communication at 13 Modules, the problem the data cable was too long between the last Module and the ESP.

I placed the ESP very close to the last module and shorten the data cable.

The long cable was no problem between the other modules only from the last to the ESP. It seems the ESP port is too sensitive or not sensitive enough.

Nice work btw emir0815 with 70 Modules. Which data cables did you use? I soldered these ones from China:
50 Sets Micro JST 2.0mm PH 2-Pin Connector Plug With Wires Cables 120mm (152844789679)

emir0815, 2021-09-15:
Hi,

 

I've used cat 6 network cable whisch has drilled wires and a metal mesh which i grounded on one side... for connection between esp and batterybanks and same between battery bank (cell1/14 with cell2/1).

 

between modules i have soledered in prefab cables with jst... as you did. meaning, cable soldered to module then plug male on female and wire to next module...

 

I haven't had time to care about it for some time due to travel and job and so on. hope this changes in coming days :-)

 

regards

Karl

 
 

Gesendet: Mittwoch, 15. September 2021 um 14:28 Uhr

stuartpittaway, 2021-11-26:
This issue should now be resolved in the latest code update with higher communication speeds.  Please test and let me know how you get on.

emir0815, 2021-11-26:
Really appreciate! Many thanks!!!!!

 
 

Gesendet: Freitag, 26. November 2021 um 10:34 Uhr

emir0815, 2021-11-26:
do I need to update controller only or modules as well...?

 
 

Gesendet: Freitag, 26. November 2021 um 10:34 Uhr

stuartpittaway, 2021-11-26:
Take a look at this video which hopefully explains all.

https://youtu.be/FUMwssfSib0

emir0815, 2021-11-26:
well done! many thanks!!
 

Gesendet: Freitag, 26. November 2021 um 11:11 Uhr

 

https://youtu.be/FUMwssfSib0

emir0815, 2021-11-26:
By updating the controller, is there any improvement for 4.21 version to be expected...?

 
 

Gesendet: Freitag, 26. November 2021 um 11:11 Uhr

 

https://youtu.be/FUMwssfSib0

stuartpittaway, 2021-11-26:
Unfortunately not, only 4.40 modules can use the faster speed.

emir0815, 2021-11-26:
oh shit! I hope jlpcb will be able to deliver in near future...

 

thanks anyway!

 
 

Gesendet: Freitag, 26. November 2021 um 13:49 Uhr

closed 2022-04-14 by stuartpittaway