stuartpittaway/diyBMSv4 #89: Communication Problem.

Neeraj-Ch, 2021-07-16:
Hi Stuart, I Ordered 4.40 Boards & ESP8266 controller module. All Populated by JLC SMT service. 
Everything worked well, from programming to everything else. I was able to find all parts. 

I connected a single 4.40 cell module  to a li-ion cell having voltage of 4.0 at that moment, for testing how BMS works. Wired TX of module to RX of Controller and vice versa. 
Problem is i am getting error "The controller is having difficulty connecting with the cell monitoring modules"
& "Waiting for modules to reply" At same time. 
The Blue LED on cell monitoring module have 2 brief flashes than a gap for about 5 secs. 

Checked for any issues but didn't find any. 
Also reversed the communication wires. But no luck
Pls Help. 
Wiring is OK. Cross checked 100 times.

stuartpittaway, 2021-07-16:
Do you see anything on the graph web interface?

When you programmed the attiny chip, did you also program the fuses?

Neeraj-Ch, 2021-07-16:
First of all Thanks 🙏 Stuart for Help. 
It been a Miracle. When i started to test after a break, all worked as charm. 
I don't know what puzzle it was. Now its working as it should. 
You are doing a great work man. Keep it up
Greetings from India. 🇮🇳.

stuartpittaway, 2021-07-16:
Ok, glad it's working.

closed 2021-07-16 by stuartpittaway