stuartpittaway/diyBMSv4 #90: Active Balancer

jcastro, 2021-07-17:
Hi!

First of, thanks a trillion for this great piece of hardware/software. I was wondering if there's any plans to design monitor modules but using active balancing instead of the current passive ones. Thanks!

stuartpittaway, 2021-07-17:
No plans, I'm not really a fan of active balance device. I can see why people think it's a good idea, but I just think they hide problems.

jcastro, 2021-07-17:
> No plans, I'm not really a fan of active balance device. I can see why people think it's a good idea, but I just think they hide problems.

Thanks for the reply! Have you ever written something about your thoughts on it? I would be delighted to know more from a person with your experience and knowledge!

stuartpittaway, 2021-08-02:
I've not written anything down, but if you take a look at the thoughts Peter Matthews has over on the DIY Powerwalls group (facebook) - I largely agree with his comments.

closed 2021-08-02 by stuartpittaway