stuartpittaway/diyBMSv4 #97: Add "Battery capacity"

magpern, 2021-08-22:
**State of charge**
For a State of Charge to work, Shunt needs to know the battery capacity.
Add a "Battery capacity" field in shunt settings.
Display SoC in main window (with one decimal).

**"Consumed Ah"** would be a nice feature.
When Battery is full, manually reset the SoC is 100% or otherwise set to "now battery is full", Consumed Ah is 0Ah
While discharing, Consumed Ah decresses, it counts down.
When charging, it counts up.
When it hits Consumed Ah 0Ah it stops counting. It will never raise above 0Ah.

**Syncronize SoC**
There should be a way to manually set SoC. If you know batteries are full, you manually syncronize SoC to 100%.
This resets "Consumed Ah" to 0Ah
Likewize, if you know SoC is 50%, "Consumed Ah" is set to half of "Battery capacity" times minus 1

**Automatic SoC syncronize**
Introduce a "Battery full voltage".
When shunt voltage reaches this value and stays above this value för "Charge detection time" (3 minutes) and current is less than "Tail current" (1%) (1% of "Battery capacity" eg. 560Ah battery when hits 56V for three minutes and it charges with less than 560*0.01 = 5.6A during this time = battery is full. "Consumed Ah" is reset to 0Ah

**Unnecessary values**
The other values "Ah In" and "Ah out", contributes with no usefull data (circled green).
[image]
Personally I don't think they belong in a overview window, more like a "extended data section"
Since they never resed, they are more a "Total Ah in/out" counter.

fhorst1, 2021-08-23:
Reset the SOC at 50% is not realistic. 
Lithium knows only full and empty.
All voltage readings in between are rough indicator, 10-20% wrong.
Reset at 0 or 100% is good.

Setting battery capacity, tricky one.
Presumed AH, agreed.
Real capacity will be known after a few full / empty / full cycles.
Where empty could trigger an capacity reset, based in the AH in till fully charged. (58v for LiFePO4 S16) 

Personally, I like total AH in and out counter.
It gives a great indication how many cycles you have done.

Also the efficiency of the lithium battery.
It should be possible to reset when the cells are fully charged.

This way, every time the cells are fully charged, the in and out should be the same, if the cells where 100% efficient.

We all know that they aren't.
At 100% fully charged, there is a known start point.

Nothing more can go in.
All that goes out can be put in again + efficiency loss 
Actually...
I would find that a really nice information, in %%.%

magpern, 2021-08-23:
> Reset the SOC at 50% is not realistic.
> Lithium knows only full and empty.
> All voltage readings in between are rough indicator, 10-20% wrong.
> Reset at 0 or 100% is good.

I agree. But if you where to do a firmware update that wipes all data or something else or have other type of battery, a manual "set SoC to what ever you want" would not hurt

> 
> Setting battery capacity, tricky one.
> Presumed AH, agreed.
> Real capacity will be known after a few full / empty / full cycles.
> Where empty could trigger an capacity reset, based in the AH in till fully charged. (58v for LiFePO4 S16)
> 

If you do full and empty, yes, when you know. But if you don't, the easiest would be to just set the nominal Ah of you batteries and go from there. It's just an indication, not actual fact.
I am compairing with Victron SmartShunt, which in a sense is the "gold standard".
With Victron you set the nominal Ah of you battery. It also keeps trach of "Lowest discharge". If "Lowest discharge" is more than nominal Ah, then you know your battery capacity is more then you entered.
It also counts cycles, life time Ah In / Out

> Personally, I like total AH in and out counter.
> It gives a great indication how many cycles you have done.
> 
It does, but in the main windo of the diyBMS webpage? It is, to me, something you look at a little now and then. It could at least be in the bottom of the page, instead of at the top. Or in a "Shunt Data" tab by it self, along with other raw shunt data?

> Also the efficiency of the lithium battery.
> It should be possible to reset when the cells are fully charged.
> 
> This way, every time the cells are fully charged, the in and out should be the same, if the cells where 100% efficient.
> 
Usually, my smartShunt resets itself everytime I reach 55.5V (it is my 100%) for x minutes and y% charge. When inverter goes into absorption, there is a few amphours left to charge, the float. At float - you should be full. And Ah should be reset.

> We all know that they aren't.
> At 100% fully charged, there is a known start point.
> 
> Nothing more can go in.
> All that goes out can be put in again + efficiency loss
> Actually...
> I would find that a really nice information, in %%.%

fhorst1, 2021-08-23:
Yes, firmware update is a good point.

People should be aware (manual) that the number entered is just indication.

Real Capacity is known after several cycles.

I would like that it could automatically update the capacity, based on a discharge run (without charge)

Even if you have 250Ah LiFePO4, and you write 200Ah, when it passes the 200, it should increment.

Perhaps different tap, real maximal used battery capacity or so :-),

I would like to see this information automatically generated.
It will provide insight how deep I maximal discharge my cells, and how much head room is available.
It should be resettable, sometimes there can be an error with solar, depletion to your max, but that's not normal max usage.

Other thing useful would be average usage per day.

Location on the page...
Perhaps new tab with "statistical data and analysis"

That would make sense for location.