stuartpittaway/diyBMSv4 #103: JLCPB rearranged some parts

wegrei, 2021-11-28:
Hi Stuart,

first thanks for all the work you have done for the community. I follow your channel since a while as I am working on a solar power wall for my house. Have collected 2500 18650 cells so far. Now I am ready to build the first bank. Did modify your 4.40 module a bit as I am working with an external Pt resistor. Skipped the break away design for my module and rearanged the Power1 receptable a bit (makes it easier for my snap in mounting). Worked all fine. Did send the gerbers, BOM and CPL to China. They have reviewed the module and changed the orientation of D2 on the board (see picture attached). Is that right?
They also rearranged some parts on the controller board. In the DFM Analysis I got a warning that PAD to PAD spacing is too large. I think I can skip that warning. Then they changed orientation of D1. This is in order with the marking on the boeard. Is that correct? Furtheron they turned the SD-Card around, which I think is the bigger problem as the screen is covering the slot then. Is there a problem in the design or did they make a mistake there. Would be nice if you could help me out.
Again thanks a lot, greetings from Germany, Wern
[image]
[image]
er

stuartpittaway, 2021-11-28:
Normally jlc know what they are doing, so it's likely ok to go with the changes.

The problem is that they keep restocking parts and at the same time change the orientation, so I. Can't keep up with their changes.

wegrei, 2021-11-28:
Thanks Stuart,

that helps. Lets wait and see what shows up.

Have a good one, Werner

wegrei, 2021-12-12:
Hi Stuart,

just received the printed boards from china. There really seem to be a problem in the design. As it looks like the SD card has to be turned by 180°, as JLPCB did. Then you can not access the card anymore as the slot is pointing inwards. Doublechecked with the datasheet, the circuit and the printed layout. Seems that the layout has to be turned by 180°. Pls check if I am right. I might have to order a new set then.

Thanks

Werne
[image]
[image]
[image]
[image]
r

stuartpittaway, 2021-12-12:
Did you notice that the card slot is the same as used in some old mobile phones? You slide the metal cover up and the card does inside. It doesn't push in from the side.

The PCB was changed to use this part about a year ago, so jlc could assemble it. The notch on the PCB for the card was also removed at that time.

wegrei, 2021-12-12:
Thanks Stuart,

the holder JLPCB mount is a fixed one (slide in from the side). However try to buy one which can be slided up. It still will be hard to get access to the card as the screen is mounted above. I have to mount the screen then with a noticable distance to the board to make it happen. Anyhow I will get it fixed.

Thanks again and have a good day

Werner

stuartpittaway, 2021-12-12:
The photo above is not a fixed one. You slide open the locking tab on the top.

The card isn't designed to be regularly removed, so no problem with the TFT screen above.

wegrei, 2021-12-12:
Oughh,

now I see. You are right I can slip / flip it open. Then it works for sure. Sorry for all the stupid questions and the confusion. Again thanks for your help.

Werner

closed 2021-12-13 by stuartpittaway