stuartpittaway/diyBMSv4 #105: diybms wifi connection issues

gjonesy84, 2021-12-22:
Hi,

Ive recently ordered the diybms controller and v4.4 cell modules. So far very impressed. Ive got it talking to an rpi with canhat to the venus os.

However I have a wifi issues. After 20mins or so the controller disconnects its self from the network. Canbus stays alive and so doe the controller.

So I have tried the following:

Made the ip address static in router.
Reflasehed another ESP32 with the very lastest firmware.
Tried a different PSU.
See below photos of my ‘testing’ set up. Screen displays its connected to ip address 192.168.0.132. But if connected via serial the messages clearly state its disconnected from the network.
[image]
[image]

Any idea why after 20min it would loose connection to network?

Any help would be much appreciated.

Cheers

Gareth

stuartpittaway, 2021-12-22:
I have not seen this exact problem before, however I would start with looking at the power supply to the controller. Can it supply 2Amps at 5V?

pihome-shc, 2021-12-22:
i had similar issue, first check power supply (try to get decent power supply not cheap chines) and second that USB cable could be the issue as well.

gjonesy84, 2021-12-23:
Morning Chaps,

Thank you for your replies. So I have changed USB adaptor to Samsung one and new USB lead and unfortunately still doing it. 

I now have a benchtop PSU connected directly to the 5v line amd still have wifi connection issues. 

Any other suggestions? 

Cheers. See set up below. 

[image]

stuartpittaway, 2021-12-23:
On the TFT screen it will have the signal strength of the WiFi, bottom right corner.

gjonesy84, 2021-12-23:
Worst I've seen it is 69db.

gjonesy84, 2021-12-23:
I moved my Mesh wifi right next to it now. Hopefully that sorts it. Now 20db on WiFi signal. 

[image]

gjonesy84, 2021-12-24:
Morning all,

Think my connection issues were a cause of few issues.
1) Reflowed the solder connections on pcb. 
2) USB lead was a bit naff
3) suffered some.strnged network/Mesh issues. Restarted everything and it's now been up for 12Hrs. Hopefully stays that way.

Cheers

Gareth

closed 2022-04-14 by stuartpittaway