stuartpittaway/diyBMSv4 #106: Feature request: 30A Buck-Boost converter per cell for balancing

renne, 2022-01-18:
Great Project! :smiley:

Burning over-voltage with resistors is inefficient and causes heat problems.
It cannot help with under-voltage of cells.
About 1A balancing current is to low for C1 310Ah cells.

I suggest to use a 30A Buck-Boost converter per cell connected to the main rail for balancing.
It can be driven via PWM-GPIOs of the ESP32.

renne, 2022-01-20:
After seeing this [video](https://www.youtube.com/watch?v=1KgcYDUvVxc&list=WL&index=120) on Youtube, it would be sensible to have a current similar to the C1 value for LiFePo cells (e.g. 300A) for real-time balancing.

stuartpittaway, 2022-01-20:
I fear that linking cells via a buck-boost converter would begin to loop/link all the cells together unless the converter was completely isolated.

How would you achieve electrical isolation between the boosts?

fhorst1, 2022-01-20:
You need to have really crappy cells to need more then 1A balance per cell.
The 250 and 400A Dali BMS have 0.08A. To give some comparison.

While burning off additional charges while the rest catches up isn't the most energy efficient way, it's power you normally would not use.

The solar charges, and balancing happen only at the top.
If there is due usage imbalance of 1A between 400Ah cell..
That would make one 399A and the other 400..
And it would take an hour to top charge with 1A to let the other cell reach 400 also.

30A is exceptional high.
Even with really crappy 456Ah (S3 152Ah) cells 1A active balancer was enough to keep things balanced.

Perhaps with speed charging and electrical cars massive balancer like this is needed.
Not for solar.

I do like the idea of active balancer (that works), a separate balancer costs about $1.25 per cell for 1A.
5A is about $3.50

It's there, it's dirt cheap, it works.
And, it doesn't bite the DIYBMS!!!
If for what ever reason the active balancer isn't capable of keeping up with the charge, the DIYBMS can burn off the top charge of the highest cells.

(And in emergency stop charging all together, where the active balancer can shift some energy between cells.)

In my humble opinion, active balancer shines at "empty" batteries.
Standard passive balance (burn off excess) is absolutely capable of keeping the pack balanced. (At top)

Whe you do need to access that last 10%... And one or more cells are dipping...
That's when active balance makes all the difference between keeping the lights on at night, or use candles.

Don't expect wonders, in low power usage modus of your home, it will make that difference....
Light or darkness...

With over 1000ah @S16 ... It won't happen fast for me.
And.... At the moment 2A active balancer does a perfect job ...

The BIYBMS is more to provide insight then really actively "work".

In your comparison... 
My +1000Ah would need 100A balancer???
2A is more then enough.
1A would have been sufficient.

If you think you need 100A...
You either not yet have cells to play with, and in theory it sounds good, or you need to test your cells as 1 or 2 are to bad to be able to use and need replacement ASAP.
Bad cells can damage the whole pack.

After initial balance, no 2 cells should have 10% SOC difference in the same pack.

If you do have problems like this, no active balancer or BMS is going to fix this problem.
Replacing the cell (s) will.

fhorst1, 2022-01-20:
> I haven't bought the LiFePos, yet, because of the Covid-19 upcharge and a
> lot of problems with detecting the capacity because of the flat
> charge/discharge voltage (see Andy's Off-Grid-Garage video link in #106
> (comment)
> <https://github.com/stuartpittaway/diyBMSv4/issues/106#issuecomment-1017586246>).
> In the video the real capacity drifts apart up to 30%. The balancers cannot
> detect the difference because of the flat charge/discharge voltage. As I'm
> going to charge or draw up to 300A from the LiFePos this means when the
> balancer detects a voltage decrease there are not much amp-hours left in a
> low cell compared to the pack. It would have to be re-balanced quickly with
> a lot of current from the pack to avoid a BMS shutdown.

Thank you Renne,

First let me point out that while I like the videos Andy makes, they are
for *ENTERTAINMENT* purposes only, and Andy is playing with Lifepo4.
He is not a professional with a licence and/or degrees in electronics, and
his findings in Australia might not be the same in, let's say, Germany or
the USA.

Second, if you want to use solar, lifepo4 for household solutions, and just
buy 310AH, and are planning to "kill them" at C1 rating...
Stay away from DIYBMS, or anything DIY. It's not built for these massive
loads.

Why on earth would you even want to draw 300A?
12v 300A would deliver 3500 watt, sure, no one in their right mind would
really want to squeeze out 3500 watt on 12v.
98% of the inverters that claim they can really pull this, will provide
magic smoke if you try to pull this for an hour..
The few brands that can pull it off... well...
Let me put it this way, if you are willing to pay $5000 for that quality
inverter, you absolutely are not stopping at spending $500 for 4x 310Ah..
(roughly $125 each)

If you want to pull anything close to 3500 watt.. you do need to go to 24
or 48v
I power my whole house, 5 x air conditioning, 2000 watt deep well pump (and
3 smaller ones 200-500 watt), 2000 watt electric oven, and all the modern
goodies.
(including 24/7 mining rig of 850w)
We live off grid. While I do have 2 x 3.2kw and one 5.5kw inverter, one 3.2
is capable of running the whole show.
Yes, we would need to do some time management for the big users, we can
(and have) run weeks of one 3.2kw inverter.

Theoretically, if we connect all the energy consuming devices at full power
consumption at once, we reach 10.000 watt
Day to day basis, we almost never go above 3000. Unless we make pizza :-)
The oven @2000watt is hungry. and while scheduled not to fill the tank at
night, the deep well pump can theoretically run at the same time, plus the
mining rig, air conditioning.

This is a house for 6 people, practically never reaching 3500 watt...
Over theorising is what I did at the start also.
I made the mistake of buying the biggest available capacity at that time,
152AH, which had the same casing as the 120 ah cells.... adding 34Ah on
sheets didn't work out as planned.
Now I have 48x 152 AH bricks and am happy that we survived the fire.
Please do NOT make that same mistake. 310Ah has the same casing as the 260
and 280Ah....
You might be in for a sad surprise.
if you want to draw 3500watt, go at least 24v, better 48v.

Asking for an 30A active balancer is trying to find solutions for a badly
constructed setup.
As you don't own anything yet, I suggest you go back to the drawing board,
and rethink your design.

Most important is that you design it in a way it will fulfil your needs.
For us, living off grid, that is having power even in the rainy weeks.
That explains the +1000Ah battery (54 kwh) and 43x 335w solar, 4x800w wind,
and how we choose to use the extra capacity during the sunny months for
crypto.

For you.. only you can tell.
If you want to be able to drain the battery in one hour as UPS for servers,
that's OK. Super fast Ebike....., sure.
At higher C rates the imbalance of the cells is bigger. And that's OK.
You probably aren't going to charge the ebike in an hour, and just accept
that you will have 80% available of your capacity.
What is also normal to use, if you want to keep your battery healthy.
Lead acid (deep cycle) has only 50%...
Most people try to keep away from the top (3.65v for lifepo4) and low
(under 2.5v) so you end up using 80% anyways.

If you do need the additional 20% it's much cheaper and safer to buy 25%
extra capacity then to move energy around to try to compensate.
2A balancers already tend to get warm... inmagine what a 30A would be
like... a car heater :-)

Lifepo4 is used for EV. buses, trucks, transport. not race cars.
If you really need to drain it empty in one hour... Probably better not to
choose Lifepo4 but different chemical structures

If you agree I think we can close this request for 30A active balancer for
this DIYBMS.
Outside "targeted area" of the DIYBMS :-)

renne, 2022-01-25:
We're going to power a fully electrified CO2-neutral household with 3 [Victron Multiplus II 48/5000/70](https://www.victronenergy.de/inverters-chargers/multiplus-ii) (using one already with AGM batteries + grid). 3 Multiplus II can draw up to 300A. This could happen for 15-30 minutes when cook hob (7 kW), oven (2.5 kW) and air-conditioning/heating (4 kW) are running simultaneously. Safety and durability of LiFePos (Lithium-Titanat are too expensive) makes them ideal with the exception of measuring the state-of-charge for balancing with the flat voltage curve.

I do have much more faith in open-hardware/open-source projects based on collective intelligence than in empty promises of commercial black-boxes. I would love to see a DIY BMS project for powering households.

renne, 2022-01-25:
@fhorst1 Where can you get active 5A balancers for about $3.50?

closed 2022-01-25 by renne

fhorst1, 2022-01-25:
:-)

I join your faith..

As I hoped to explain, 300A is not the problem.
Drawing 300A from 310Ah cell is.

With 4kw airconditioning, besides the cooking...
It's for sure you aren't going to drain 300A from 310Ah cells
Simply not enough energy storage left over for the night.

20 minutes cooking, at 7000 watt. 2.2kwh
2 X 15 minutes shower, 5000 watt
2.5kwh
Airconditioning 1500 watt for 12 hours 14 kWh
All the things I missed....
Say 20 kWh needed per night before the sun is strong enough to recharge.
48v, that's S16 310Ah
16x3.2x310= 15.8 kWh

Not even keeping in account the rainy days.

I would advise to minimal use 3 sets, and not drain 5h4 cells each night to
damaging low levels.

With this setup, I'm quite positive your AGM is around 1000Ah, at least.

With 3 sets, you drain not C1 but C0.35
At this level (300A), imbalance stays way within parameters, where an 1 or
2A active balancer is powerful enough to keep in pace.

95% of the time you are draining 2000 watt or less (probably less)

Without my crypto mining, we do about 750-1000 on average.
That's including the 2000 pumps, Airconditioning, lights.
Thailand, 25-30 at night, 35-40 daytime with peaks to 45-47.
Celsius

Yes, we don't cool the whole house 24/7.
That would need some more solar, or no crypto :-)
2 roooms are always at 25.
The rest at 30.
And real good insulated walls and roof do the rest :-)

With 3x 310... You already might get into troubles on a rainy week.
Perhaps take the jump to 4 sets.

I have tried to parallel...
It makes the world a lot more complicated.
You are absolutely free to play how ever you like.

But will end up to the same conclusion that makeing 4 individual sets, each
with own controller, and 48x DIYBMS cell monitors will work the best.

All hooked up to the backbone that you have your 3 victrons hooked up to.

I choose JiKong Active balancers, as they have BT, and provide nice
additional information.

They are high quality, only negative is that you need to be within 10
meters to read out the balancer.

Many cheaper solutions available, and most of them work.
Please put them in an enclosure/box/case as the capacitors do sometimes pop.
You don't like that close to the cells....

Let us know how the setup progresses!!

Frank

fhorst1, 2022-01-25:
> @fhorst1 Where can you get active 5A balancers for about $3.50?

5A active balancer for $3.50 per Cell...
S4 is $14
S8 $28
S16 $56

Jikong 1A about $80, 2A $120, 5A $350 they even have 10A for $550 or so.

https://a.aliexpress.com/_msWLku4

I'm not affiliated in any way, this is just one of many sellers.
With some searching you can find cheaper ones.
My 2A costs me $110 include transport to Thailand.

This type works (I tested the S4) 
https://a.aliexpress.com/_mLJs7ic
1.2A (max) is about $4.15 for S4
(Bought locally)
My gut feeling doesn't thrust it to continue to work for 10 years...

This type works also, I did not test myself, a friend did...
https://a.aliexpress.com/_mKU0V80

He had 2 different sets, one a capacitor popped after a month.
Other one is working for 15 months now.

Both sets are same balanced...
One with, one without active balance..
(Daly 250A BMS)

renne, 2022-01-25:
@fhorst1 
I just found [this](https://de.aliexpress.com/item/1005003132170432.html?gatewayAdapt=glo2deu&spm=a2g0s.8937460.0.0.61ac2e0eNcRKyf). It looks like multiple bi-directional DC/DC converters connected to a common coil.

fhorst1, 2022-01-25:
40 euro's for 10A seems to good to be true.

Sadly... It usually is, but for 40 euro...
Worth a shot.

Personally I would use 4 small 18650 lithium cells, and drain one, hook it
up and see how it goes.

If it does work....
Good find!