stuartpittaway/diyBMSv4 #107: O. O. S and C. R. C errors

DCGenerators, 2022-02-17:
Hello.
New here!
Having the following issue with a newly installed setup of 16 * V4. 4 modules.
Trouble communicating with modules with OOS and crc errors.
Controller is an esp32 nodemcu
[image]

stuartpittaway, 2022-02-17:
Ok, let's break this down a bit.

CRC errors are corruption in the signal between the controller and/or modules.

The OOS errors occur when a request is sent to the modules but no reply is ever received.

Judging by the bad packet count errors in the screen shot, I think there is a problem with the first module.

Can you swap that module with one of the others, see if the problem follows?

stuartpittaway, 2022-04-14:
No response, so closing the issue, please feel free to open if you still have problems.a

closed 2022-04-14 by stuartpittaway