stuartpittaway/diyBMSv4 #112: Does the diyBMSv4 need a capacitor pre-charge for the Victron Multiplus II / Quattro II ?

HerrFrodo1, 2022-03-28:
I've found that pre-charging the large capacitors on the Victron Quattro II makes a lot of sense.
You can easily do it by connecting a small resistor with 22-27 ohms, 1-2 watts in parallel to the contactor with a button. Press the button for a few seconds, the capacitors are pre-charged. If you then connect the battery cables, there will be no more sparks.
Now comes the problem!
You can't always stand by and manually precharge the capacitors when the diyBMS opens the contactor due to a rule and then closes it again at some point.
The contactor then has to withstand many 100A (which the battery can handle). My contactor TE LEV200A5NAF 900V/500A) now has a permanent passage - it no longer opens.

My suggestion:  You need 2 relay outputs on the ESP32, which can be set slightly differently.
Relay 1 must be configured as a fleeting contact relay - means it must turn on and close again after an adjustable time. 
If relay 1 is then closed, relay 2 must switch on and switch the contactor. It would also be possible to program relay 2 as a time relay with response delay.
Whenever the controller is started or a rule causes relay 2 to switch off, the aforementioned procedure must be followed.

What is your opinion?

Best greetings from bavaria,
Jochen

fhorst1, 2022-03-29:
Answer simple:

No it doesn't.

And yes, almost all Inverter do benefit from pre-charge with a capacitor.

Not DIYBMS or Victron related.

It's a good tip that is shared regularly on YouTube, Will Prowse and others.

Some inverters have a system build-in to prevent a spark or a high peak at
start of contact.

I had an 150A Daly smart BMS who needed pre-charge inverter, otherwise it
would not even function.
(Went in protection due the high peak directly)

And a dumb 250A Daly who didn't.

Like most equipment, peaks aren't the best, and sparks neither.
Prevention is better and using a capacitor to pre-charge an inverter is a
good thing.

Not especially for DIYBMS nor Victron.

Thanks for the tip 👍🥰🙏

HerrFrodo1, 2022-03-29:
Dear Horst,

thank you for the tip too. 

I hope Stuart and his friends can program it. Maybe he has another good idea.
Yesterday Amazon brought me 2 adjustable time relays, I will test them one day.
I also have to replace my defective contactor. I discovered an interesting contactor here in Germany. 
If it's good, I'll let you know.

Thanks and regards,
Jochen

stuartpittaway, 2022-03-30:
This could be a useful enhancement for the BMS, and not that difficult to achieve.  

The BMS would need to have a new "start up" sequence or feature added.

HerrFrodo1, 2022-03-30:
That would be great! Thanks!
On the hardware side, you then need an additional 48-60V contactor (for 16S) with the same coil voltage as the main contactor - only significantly smaller. Are there any discussions about which contactor are good?

stuartpittaway, 2022-03-30:
If you are only drawing a watt or two during the pre-charge, then any suitable relay could be used, you would just extend the duration of the pre-charge depending on the size of the resistance used.

Michi202020, 2023-12-21:
Hello i need also the timing for the Pre Charge Relay.

can you help me withe the Pre Charge. I have Build many Batteries and the Relay Controlled withe a seperate controller. How can it config withe Relay1 and Relay2?

stuartpittaway, 2023-12-21:
Hi @Michi202020 pre-charge isn't currently supported in an automatic way by DIYBMS - I simply wired up a push button + high power resistor for the rare occasion I need to disconnect the battery.

Michi202020, 2023-12-21:
Hello Stuart,

In my case I have install in my family many system and for example the Diybms have an error and after many yours he reconnect than I am not there and my onkle have no idea from electricity. In this case I find it to risk. Is it a big problem 1 timer to used for the delay the second relay.
Best Regard
Michael

stuartpittaway, 2023-12-22:
> Is it a big problem 1 timer to used for the delay the second relay.

A precharge timer would need some code changes, wouldn't be difficult, but not currently supported.

HerrFrodo1, 2023-12-22:
Hi!
Here is my solution for preload Victron MPII or Quattro II or whatever :-)

[Preload-Inverter.pdf](https://github.com/stuartpittaway/diyBMSv4/files/13753445/Preload-Inverter.pdf)

Michi202020, 2023-12-23:
Hello Stuart,

If you can make it than I find it great. I can also give you per PayPal for your work. I am also patron but this work I think it's more for me.

Michi202020, 2023-12-23:
> Hi! Here is my solution for preload Victron MPII or Quattro II or whatever :-)
> 
> [Preload-Inverter.pdf](https://github.com/stuartpittaway/diyBMSv4/files/13753445/Preload-Inverter.pdf)

Hy HerrFrodo1,

See my Picture I install it all in a box for better quality and easy installation. Your idea is goot but to much components. Many thanks. 
[image]

Michi202020, 2023-12-23:
[image]
[image]
[image]
[image]
[image]

Michi202020, 2023-12-23:
[image]