stuartpittaway/diyBMSv4 #113: Balancing with the diyBMSv4 with CANBUS to Victron Multiplus II / Quattro II and 16S1P EVE LiFePo4 230Ah cells

HerrFrodo1, 2022-03-28:
Hello diyBMSv4 fans and those who want to become one :-)

First of all you need to know my setup:
DIYBMS ESP32 release 26.02.2022, 16x Module V4.4 (9,6K) - Roundtrip: 880ms
2x 1m twisted cables to modules on cells. 8-10cm cables module to module.
DC/DC power supply 48/5V Meanwell, UTP cable Victron CANBUS to Victron Cerbo GX
DC/DC power supply 48/24V, Contactor  LEV200A5NAF(broken)
Victron Cerbo GX (master unit between Victron Quattro II, diyBMS, smart shunt 500A and more)
70mm2 Batterie cables from Batteries to Quattro II. Smart shunt in minus and 300A fuse and contactor in +.
The Cerbo, contactor coil, diyBMS gets their battery power after the shunt and before the contactor. 
Each one has a circuit breaker for DIN rail.
The Victron Quattro II has to be setup by an MK3 USB adapter for general settings. After that you can view and make some settings with the Cerbo GX on a webserver - I use LAN for it. 
The diyBMS communicates with the Quattro II and controls the charging power. Some diyBMS values are displayed in the remote console and in Victron's VRM portal - I only have a smart shunt from Victron (at the moment)

The Victron Quattro II is identical to the Victron Multiplus II (best-selling) - the Quattro II only has one more AC input (e.g. for a generator)
The ESP32 transmits the CVL, CCL and DCL via the CANBUS over the Cerbo GX to the Quattro II.
The Quattro charge with these values. I can confirm that. 
But the Quattro charge according to its own charging curve. 
As soon as a battery cell enters the bypass threshold area, the diyBMS transmits the balancing data (CVL, CCL, DCL).
So far so good. The only problem is that the Quattro then does not reduce the charging power quickly enough. There is no middle area - something like a few mV before... the point of balancing is the last one.
At the moment it's not bad because I balance to 3.52V. I had tried it with 3.65V and then it is dangerous for the cells - especially since the balancing current is not sufficient. I connected an active balancer, so it's much better.
The BMS always tries to balance exactly on the bypass threshold value, so it must be the last step. 
If it balance earlier, then all cells go into full balancing mode and heat up - the Quattro then doesn't know the right energy to bring each cell to 3.65V. In addition, the BMS tries to reach the lower value and a lot of energy is lost. That's why I've now set the bypass threshold to 3540mV and the maximum CVL to 56.4. This value is better for the cells and brings more charge-discharge cycles. Not much should be lost in terms of capacity. 
The Quattro now charges up to 56.4V with full power (what the cells are ready to absorb) and as soon as a cell reaches the bypass threshold of 3540, the Quattro switches to balancing mode via CAN from the diyBMS. Now the performance is reduced and if cells are now well above 3540, that's not a problem because the cells are still far away from 3650. Then the end-of-charge voltage of 56.4V is reached and the Quattro regulates down to 0.1A or even 0A. Now the BMS balances much more relaxed and the resistors don't get hot.
I flashed the ESP32 and the V4.4 modules to the latest firmware. In addition, the modules to 9.6K. Roundtrip 880ms, before 3300 - that's great :-)
There doesn't seem to be any glitches on 16S. The bus cable from the controller to the first and last module is about 1m long and twisted. the cables between the modules are about 9-10cm long.
All in all, the communication from the BMS to the Cerbo/Quattro is ok if, like me, you first balance at 3540mV and set the maximum charging voltage to 56.4V. If you want to get to 3.65V per cell, then we still have to work on how to do it better.
In order to be able to balance more passively, you have to change the whole board or plug/connect an additional balancer board to a DIYBMS module. That is certainly a lot of effort and you burn more energy unnecessarily. 
Either you develop a separate active balancing board or you take one from China.
I think if Mr. Pittaway should add another line to the Victron settings and call it "approach voltage".
Absolute end-of-charge voltage is 58.4V and 3.65V per cell (LiFePo4) - you have to balance on that - but you have to balance from below 3.65V, if the cells won´t get over 3.65V.
So first set a total voltage up to which the Quattro will charge full power and then one from which it will reduce. If the Quattro has then reduced the power and the cells are balanced to eg 3550mV (+/- a few mV), then continue charging with low energy (0.2-0.5A - depending on the capacity) and the cells up to high 3.65V/cell charge further.  LowTopUp :-)
Is that possible?

I enabled DVCC and took a screenshot of my settings.
If you activate LiFePo4 in the Victron settings, a fixed charging curve is automatically set. Maybe I can change it - I have to study it.
In addition, I entered the maximum charging voltage of 56.4V in the DVCC. I had experimented with the charging current, but it didn't work properly.
The data from the DIYBMS seems to have priority. Nevertheless, he uses his charging curve - I have to do more research on that.

Whats your opinion?

Greetings from bavaria,
Jochen

fhorst1, 2022-03-29:
Good morning Jochen from Thailand!

Quite a long story to read.

To summarize, you found that the charge settings of the Victron isn't
matching your LiFePO4 cells and balance.

And you are correct.
Not negatieve intended, you need to research more.

Balancing should start at 3.5 or earlier. 3.4, 3.45v.
3.65v is the maximal adviced voltage.
At this point it's 100% charged.
That doesn't mean that as soon as it reaches 3.65v it's fully charged.
As during charge, voltage is higher than actual voltage.
(Depending on the amperage used)
LiFePO4 have amazing wide margins.
It can be charged up to 4.2v without damaging, and discharge 2.0v without
damaging.

Many are aware of the advised 2.5 as normal low setting and 2.0 as
"absolute" low point.

But not aware of the advised 3.65 and the "absolute" top of 4.2.

Going beyond 3.65 during a charge cycle won't damage your cells, unless you
push it too far.

Probably many will disagree, those need to research more also :-)
Fear based knowledge isn't knowledge.
Facts are.
A little journey on Google will educate.

For LiFePO4, longest lifecycle is obtained by charging up to 90-95% and
discharge down to 10%.

Say use 80% capacity.

Yes, it would be nice if the charger (usually solar) would slow down near
the top.

Realistically?
They usually don't.
LiFePO4 isn't lead acid, and the reduction in absorbing the energy is much
steeper curve then lead acid.

LiFePO4 get stuck at 3.35 - 3.45v, depending on the amperage and brand of
lifepo4.
It seems to hang there during charge "for ever", up to the point the cell
has 90% (95%?) absorbtion.
Then the last part goes fast.
After the "knee", it can be minutes till the 3.65 us reached and passed.

As most chargers don't listen to individual cell voltage, it can't reduce
charge capacity based on this.

With a result that the BMS needs to step in, and stop the charge, burn of
the top, so the rest can catch up.

A well balanced set hardly ever has this problem.
All the cells reach the knee roughly at the same time, and the overall
battery voltage is at a that high level that the charger does reduce its
amperage.

Even a Daly BMS with just 0.08A balance (burn off) capacity is able to keep
up and balance the cells.

If the DIYBMS isn't powerful enough with +/- 1A capacity...
You can increase with adding a heatsink.
Even add 2 or more cell modules at the "acting up" cells.
[... 108 lines cut ...]
Stop-start for many many days :-)

That is with 1% imbalance at 280Ah.

I have 4 sets of 260Ah.
One Eve, 3 BYD.
With active balancer BMS (jikong) because I like the way it works, and 4 X
DIY BMS who acts as additional safety and information provider.

I have 80 cell modules, a few spare.
64 in use.

After playing with LiFePO4 for several years, living off grid, being fully
depending on it, and have a pile of China (and western) crap of defective
or less functional BMS, I know how important BMS is, and how easy they
don't work when needed.

I have over 30 dead cells to prove this (bye bye $$$) where chargers
doesn't stop at their max 58v, or BMS don't stop at 3.65v or combination.

Victron probably would not have charged beyond 58v.
EASUN ... They do..
(don't buy)

I have some experience :-)

HerrFrodo1, 2022-03-29:
Dear fhorst1,
thanks for your short summary.
Maybe it's because of my bad English, there's so much writing around it :-)
Thanks also for your tips - I really have to google the 4.2V - I assumed that lithium polymer batteries have this voltage level and LiFePo4 at 3.65V. EVE also writes a cut-off voltage of 3.65V in its technical data. Maybe they want to play it safe.
I had written that I set the diyBMS balancer to 3.52V per cell and 56.4V pack and in the rules cells cut-off at 3.65V.
So it now works well that the diyBMS controls the Victron Quattro.

[image]
[image]
[image]

Best Greetings, 
Jochen

fhorst1, 2022-03-29:
Hello Jochen,

Just to be clear, charging to 4.2V is NOT recommend.

It will not damage the cells, if you have played some with lifepo4, you
know how fast it can spike after 3.5v.
Daly has set their cut off voltage on 3.75, already taking more "risks" as
the standard 3.65.

It's NOT recommend to keep the cells at 4.2v, and this WILL damage the
cells.
Keeping them at 3.65 will not.

Stopping charge at 3.65 will set the cells to recover in a few hours down
to  about 3.4v
Keeping the cells at 3.65 for several hours will provide "full" absobtion
at 100%.

Difference between 3.65v and 3.4 in rest is just a few % if it is even a % .

Due the possible risks involved of overcharge 3.65 is the safe goal.

For lithium ion, that is 4.2v, where over 4.5 is damaging.
4.4 cutting it too close...

One of the reason they say lifepo4 is the safest as it can be "abused"
without too quickly damage.
Unlike lithium ion.

Same goes for "over discharge".
Drain the cell to 2.0v and it will recover back to like 2.3 -2.4v
Drain it several times more to 2.0 it will stay at 2v.
Not healthy, also not devistating.
Drain it to 1.5v several times.. and keep it like this for a few days...
You can buy new cells :-)

The time  / capacity between 2.5 and 2.0v is like nothing.
Between 2.0 and 1.5 even less.
Uncontrolled power consumption can drain to far.

Safe side...

Stop at 3.65 -3.75v and below 2.5v
Pushing it, as you absolutely need the extra 2-3% and are in a hurry to
charge...
Charge up to 4.2v will help.
Don't keep it there, stop the charge and let it settle...
End will be between 3.5 and 3.6V

But why would anyone want this?
Especially on Solar?
Where it's not about speed or maximum capacity, but lifecycle management to
max...

99.998% won't need it.
It is useful information to know where the limits are.
Your car CAN drive 210 km/h but in the city you would be nuts to drive that
fast.

So you don't freak out when you go speeding at night, empty road, for a
short while and drive 160 :-)
(Ring road, tunnel, safe place)

Same for the cells..
Don't freak out when you reach over 3.65v.

Just don't keep them there long term :-)