stuartpittaway/diyBMSv4 #120: How to reset wifi settings in esp8266 controller

maryushr, 2022-06-12:
I changed the router and now esp8266nontroller does not want to connect to the new one, how do I reset the wifi settings in the esp8266controller, it is nowhere described.

stuartpittaway, 2022-06-12:
Connecr the two pins together marked wifi reset.

stuartpittaway, 2022-06-13:
Did you reboot the esp after joining the pins together?

maryushr, 2022-06-13:
Yes, I reset by removing the USB plug and reconnecting, then the controller reconnected to the old router.

stuartpittaway, 2022-06-13:
Ok. You could reprogram the esp chip, that should clear it's settings

stuartpittaway, 2022-06-14:
Ok.  Looking at the code, this is the process you need to follow:

1. Remove jumper from WIFI RESET PIN
2. Reset ESP8266 by pressing the reset button (EN)
3. WITHIN 3 seconds put the jumper on the WIFI RESET PIN

This should then clear the WIFI settings

maryushr, 2022-06-14:
Great, now it works. Thank you very much.