stuartpittaway/diyBMSv4 #121: JLCPCB V4 Controller Issues

robanderson, 2022-06-25:
I tried to get the V4 Controller made at JLCPCB and they have replied with a question below.

Any advice would be appreciated.

~~~~~
Hi Sir or madam,

Well got your order with many thanks~

Sorry to bother you, but there is an issue that we want to confirm with you before proceeding.

As shown below, we didn't find 1.152mm holes for tooling holes, can we add it by ourselves per our rule?

![image](https://user-images.githubusercontent.com/907070/175763781-8f4e1199-d799-4978-a806-0847bb426c21.png)

robanderson, 2022-06-27:
If the solution is to just let then put the holes where they like, as I suspect then a follow up confirming from someone would be appreciated.

stuartpittaway, 2022-06-27:
Hi @robanderson, yes the solution is just to let them do the holes.

closed 2022-12-13 by stuartpittaway