stuartpittaway/diyBMSv4 #123: Made new PCB with 14s on it

N1c084, 2022-08-20:
Hi 
What do you think about to made one bank PCB (14s ) with only one tx/rx cable and link between each -/+ directly on PCB, like this we need less space and wires ?
See pic with only 2s for now 
[image]

N1c084, 2022-08-20:
Green wires are old design connexion for middle point.
Normally it would work nice without these 2 wires until it is a middle point the white is sufficient  .

stuartpittaway, 2022-08-20:
As long as you don't share the power wires between the modules it should be ok.

N1c084, 2022-08-20:
You mean voltage drop in wire when balancing will cause false reading to next cell on same wire?

>> for 1 meter wire 1mm²
R=0.0175ohm 
U=23mV for 1.3A ; ok i never think this value will be so high for so small wire/current

stuartpittaway, 2022-08-20:
Correct.

N1c084, 2022-08-20:
so i need to remove the link on PCB too, because it will cause current sharing in both wires

closed 2022-08-20 by N1c084