stuartpittaway/diyBMSv4 #125: 38S1P configuration support

Deftaudio, 2022-10-03:
Hi, I'm using 19 battery modules from Leaf for EV motorcycle project. Is it possible to configure diyBMS to support this configuration in future?

stuartpittaway, 2022-10-03:
Yes, if you search on the open energy monitor forums, people over there have built a version specifically for the leaf cells

Deftaudio, 2022-10-03:
Thank you, did you mean the redesigned module? https://community.openenergymonitor.org/t/diybms-v4/11292/550
but how to overcome the limitation to run 38 cells in series?

stuartpittaway, 2022-10-04:
The controller supports at least 100 modules (in series).

The issue you may run into with those modules is the speed of the communication. It may take several seconds for the data to pass through a long chain of modules.

It should still work though

closed 2023-01-03 by stuartpittaway