stuartpittaway/diyBMSv4 #129: diybms  bypass heating function

gjonesy84, 2022-12-02:
Afternoon,

My powerwall is in situated in an outdoor building where it gets close to ambient temperatures and could do with somekind of heating. 

It would be great if it had the software would allow a heating function. My other powerwall with batrium BMS has a heating function built in. See below

https://wiki.batrium.com/en/toolkit/control/thermal

Would be great to see this in the diybms :)

Cheers

Gareth

fhorst1, 2022-12-02:
While I see the additional protection as a plus, i don't feel it should be
inside DIYBMS.

Many commercial available heating pad (terrarium, or for hatching eggs)
have temperature settings.
It just needs power :-)
And doesn't do anything above the set temperature.
Terrarium pads are a few bucks on AliX, energy efficient, flat, thin,
reasonable flexibele, "water proof" (not for submerged)  the list goes on.
With this option widely available, cheap and reliable...
And the safety of the DIYBMS temperature setting if it gets more cold then
the heating pads and insulation can deal with...

If would be different when we building DIY heating pads with nichrome
wires, each wire being controlled by the cell module.

(I made some dump loads for my wind turbines, easy enough to make for that
purpose)

Nichrome isn't that easy to work with, especially for mild heating...
Bread toaster, your grill and oven..
But i doubt it is something you like in the powerwall :-)

Simple terrarium heating pads...
5 watt, €2,-
20 watt €4,-

Its hard to compete with those prices:-)
(I had a few in use to keep my grow tent on high enough temperature in a
too cold environment. They worked fine 3 winters.
Then I moved to Thailand, so no longer need.
Top "wish. Xzy " quality :-)

Just a thought :-)

stienman, 2022-12-02:
I'm using the older esp8266 based controller, but it had 4 relay outputs,
and I thought it had the settings to support this sort of need.  Of course,
this could be used to shut the battery off to prevent freezing related
damage during charging, but you could certainly turn on a heater instead.

Is that not the case with the ESP32 version?

As far as whether it belongs or not - thermal control for safety and
battery longevity is absolutely part of battery management, and belongs in
DiYBMS as much as anything else.  At minimum it should include the
capability of disabling charging below a set temperature, and limiting or
disabling discharge below a set temperature, if it doesn't include active
thermal management features it should at least prevent damage if the
temperature is too low.

-Adam

fhorst1, 2022-12-03:
Yes, thermal management in the form of a shutdown is and should be
available in the DIY BMS.

There is one rule set available for this, a d can be used to disconnect the
charge / discharge depending on your temperature setting.
(High and low)

There isn't a third option, so one needs to choose.

4 Relay, easy enough to use one to start giving electricity to a (heating
pad.

Using the BMS to control the temperature,  that's a whole different
situation.

Yes, you could use the minimal and maximal temperature setting to control.

But at the costs of the freezing and overheating control.
Who are in my opinion far more important.

If there is enough storage space to program additional rules?
I don't know.

I haven't found heating pads that don't have their own temperature control.
I'm sure they will be available at some location, or can modify the
existing ones.

The question was not so much if the DIYBMS can/should be able to protect
the too high and too low temperature for the battery, that it can.

The question is if the DIYBMS should be able to control the temperature
range, by powering a heater or cooler system.

Short answer, why?
All commercial heaters and coolers come with temperature control.

If it isn't broke, don't fix it. / Don't change a winning team... (And more
of this)

Why would you bypass the engineering from the company that you buy the
heating pad from, tested and tried and guaranteed solution, only to be able
to set a temperature yourself?

If you don't want to re-engineer the heating pad and loose warranty, I fail
to see the point.

They don't start or use electricity when the temperature is above the
setting.
They gradually increase/ reduce the wattage to stay at the set temperature.

You can and should use the temperature settings in the DIYBMS to protect
the battery from overheating and freezing.

Prevent freezing is also something that should be done if your environment
asks for it.
(Or active cooling when you live in the desert or so)

On the question if the DIYBMS should be the unit that controls the
preventive measures....

Unless the DIYBMS provide scalable solution that does this... I don't see
the need.
Different solutions are needed for 4,8,16 or larger Battery packs.
Then there is the size / mass, the insulation for the battery pack and the
temperature range in the environment.

I've witnessed -36 one time in Poland...
That needs a total different solution (wattage)
High powered heating brings it's own challenge...
Bypassing that tried and true temperature management because it needs to be
done by DIYBMS...

And it it doesn't needs to be controlled by the DIYBMS, then what is the
additional value of being able to turn on the heating pad a few degrees
warmer before it turns itself on?

Over complicating things will make additional points of failure.

The freezing and overheating protection is already available.
Just not to control prevention.
And the question is if one would really want to do this.

stuartpittaway, 2022-12-05:
The existing rules for "Cell under temperature (external) °C" can be used to drive multiple relays/outputs

So for example, at 0°C, the "discharge" relay could be disabled, and a "heating" relay be switched on.

When the temperature rises to the "reset value", say 5°C, the rule de-activates - so heating stops and discharge resumes.

stienman, 2022-12-05:
I thought that was the case, but it's been awhile since I opened the
configuration of my system.  Thanks!

-Adam

closed 2023-01-03 by stuartpittaway