stuartpittaway/diyBMSv4 #130: Component placement issue from JLCPCB

drawlusyk, 2022-12-13:
Hello, I placed an order with JLCPCB for boards and component placement. 

Please see the included images which were provided in my order which they fixed.
[image]
[image]
[image]
[image]

Also, the following email content was sent after my order...

****************
Please kindly note for The Standard PCBA, if the original design has no edge rail, we will add 5mm width edge rail on 2 sides of board. So the board size will increase which will lead to price changes.

Since the difference is small, we will not charge the extra fee for you this time, please kindly pay attention about this for the future orders.
***********

stuartpittaway, 2022-12-13:
Just to add I've noticed they have INCORRECTLY rotated part D2 (diode) on the modules.  If you have already confirmed the order with JLC - DO NOT POWER ON THESE MODULES until you remove and rotate all the D2 diodes.

The + should be on the right, next to the "K" mark.

drawlusyk, 2022-12-13:
Ok. I had to place an order for the missing part ATTINY841-SSU. At the same time of placing this order, I also ordered the pcb's. They thankfully sent an email letting me know that the parts will take a few days to arrive, so they cancelled the PCB order for me. Once parts are in, I will re-order with the updates. Thanks.

stuartpittaway, 2022-12-13:
Right - the new modules won't be using the TINY841 :-)

drawlusyk, 2022-12-13:
Huh? Did I upload the wrong files? I had to order 50 of these at $2 each since they didn't have them in stock.
Before ordering i did a git pull on the dir i have diybms files in. 
This is what i uploaded... diyBMSv4/ModuleV440/export/ModuleV440_bom_jlc.csv

Or are you changing components for v4.5?

stuartpittaway, 2022-12-13:
Yes, changing components - the TINY841 is difficult to get hold of, often out of stock.

stuartpittaway, 2022-12-13:
I've just uploaded the v4.5 files, feel free to use those if you want - but they need ATTINY1624 chip.

drawlusyk, 2022-12-13:
Thanks Stuart. Will upload these to JLCPCB and let you know if there are any problems.

drawlusyk, 2022-12-13:
JLCPCB is replacing the ATTINY1624-SSU-ND with is ATTINY1624-XF. Is this ok?
Still need to pre-order these but cheaper;)

N1c084, 2022-12-13:
Not the same pitch, you need to change from soic to tssop

stuartpittaway, 2022-12-13:
Did you see the video I just released? Shows the difference between the two.

You could always add the attiny chip on yourself (manually solder). Not difficult and likely lower cost than jlcpcb

closed 2023-01-03 by stuartpittaway